slic3r/Slic3r #4401: [Feature request] Transformation matrices to keep history of mesh changes

Oekn5w, 2018-05-05:
### Version
1.3.0 dev

### Problem
Reload from disk dismisses non-Z rotations and translations applied in the settings menu, as these are applied to the mesh itself without any history apart from undo stack (not viable to read from in reload)

### Suggested adaptation
* Create class to contain a transformation matrix and the required math operations
* Assign one matrix to each mesh, volume and object, maybe ModelInstance
* Changes in various parts of GUI: Instancing, PartObjectDialog
* Changes in backend: Undo/ Redo, ModelObject, ModelVolume, TriangleMesh

Oekn5w, 2019-07-18:
@lordofhypens I'm almost done with the implementation but I wonder whether I should also adapt the C++ GUI especially since I don't really know how to build with that part.

lordofhyphens, 2019-07-18:
You can probably leave it be as I don't have the part editor implemented yet.

Oekn5w, 2019-07-18:
Alright, I'll wrap it up and open a PR soon(TM)

closed 2019-12-02 by lordofhyphens