slic3r/Slic3r #4402: Z value feature suggestion

EitanSomething, 2018-05-06:
Update Suggestions

**Operating system**
windows 10

When adding a file to slic3r you don't know if the object that you what to print is taller then then the z value print space on your printer.There should be a way to set the z height of the printer under "printer settings".

EitanSomething, 2018-05-06:
#

lordofhyphens, 2018-05-06:
You can already see what the model size is in Z, and the preview will also tell you. Slic3r does not enforce the build area limits either.

I see this as extra hand holding that will annoy more users than it helps.

Very unlikely this will get implemented unless you do it yourself and in a way that other people agree isn't intrusive.

EitanSomething, 2018-05-06:
You could have it so when the object is larger than the build volume a warning will come up on the side saying so but doesn’t interfere(only a suggestion).

gege2b, 2018-05-09:
maybe just draw the build volume as a cube instead of a plane
no message, no warnings, no annoyance, only a visual feeling on how much the model is out of volume

Still needs to set the printer's height somewhere, through

Tuxomx, 2018-05-11:
Hi all. I think a good place to set the printer height is, of course in printer settings/size_and_coordinates, and add a check for "ignore this". Then if "ignore this" is set, the complete gcode is generated, and if unchecked only the gcode inside volume is generated. This, I think is important, because there are some printers which are not aware of their own Z size.

In addition, in preview and layers views, a different color could be used so the used can spot easily if something is out of the printing volume.

And, in 3D preview, a label on top of slider with the height could be useful too.

EitanSomething, 2018-05-14:
Maybe if the object is out of build size the build volume markers will turn red on the side where this is an issue.

mdmbc, 2018-05-24:
This could be very handy on a delta, where the top of the max print area is conical.