slic3r/Slic3r #4404: Building Slic3r - Building and testing Slic3r-XS-0.01 Problems

Annaf12, 2018-05-07:
Hello,

### Version
'1.302136'

 -dev version
1.2.9-1316-g8a93fbbd

### Operating system type + version
Windows10 Education 64Bit

### Behavior
I'm trying to build slic3r with Visual Studio 13 on my Windows10 Computer. I went through the instructions completely (https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-Windows) and now wanted to build the Slic3r and get the .sln, but it always fails. 
I tried the build with boost librarys 1_59_0 and also 1_60_0, but it doesn't work. 

Here my log file:
cpanm (App::cpanminus) 1.7044 on perl 5.024000 built for MSWin32-x64-multi-thread
Work directory is c:\wperl32d\cpanm/work/1525687241.10788
You have make "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"
You have LWP 6.33
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/src-perl/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4224)
Configuring C:/src-perl/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.35)
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have ExtUtils::ParseXS 3.35 ... Yes (3.35)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\wperl32d\cpanm\work\1525687241.10788\build.log for details. Retry with --force to force install it.

May someone help me? What should I do?

Thank you very much.
Best regards.

lordofhyphens, 2018-05-07:
You need to look at the log to see what is going on, as the error is helpfully trying to tell you.

C:\wperl32d\cpanm\work\1525687241.10788\build.log

Annaf12, 2018-05-07:
Hello,
Thank you for the feedback. In the logfile is written only:

cpanm (App::cpanminus) 1.7044 on perl 5.024000 built for MSWin32-x64-multi-thread
Work directory is c:\wperl32d\cpanm/work/1525687241.10788
You have make "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"
You have LWP 6.33
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/src-perl/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Checking if you have Module::Build 0.38 ... Yes (0.4224)
Configuring C:/src-perl/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Module::Build 0.3601 ... Yes (0.4224)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.35)
Checking if you have Test::More 0 ... Yes (1.302136)
Checking if you have ExtUtils::ParseXS 3.35 ... Yes (3.35)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\wperl32d\cpanm\work\1525687241.10788\build.log for details. Retry with --force to force install it.

It's the first time I try to get source code through github. I've tried several times to rebuild the installation as in the instruction described, but it always fails after I built the boost libraries and then insert
cd Slic3r
perl Build.PL 
Unfortunately, the logfile does not tell me much either. 
[image]
[image]

lordofhyphens, 2018-05-07:
@Annaf12 VC++ looks like it is trying to build in C++03 mode (not C++11).

You can change the switches in xs/Build.PL to incorporate whatever you need to tell VC++ to use C++11 syntax, or move to a newer version of VS that does it by default.

lordofhyphens, 2018-05-07:
@Annaf12 I don't do any of my development on Windows unless at great need. When I do need to work on Slic3r in Windows I use MinGW.

However, your initial error messages  (always read from the top down) tell me that your compiler is interpreting the source code as if it was pre-c++11. This is not supported.

Annaf12, 2018-05-07:
Hello, 
Thank you for the feedback. So if i can use a newer Version of VS (VS2017) how should I change the #CCTYPE in the Makefile in perl-5.24.0\win32\Makefile? At the moment  I m working with VS2013 Professional and set CCTYPE = MSVC120.
What about the boost librarys? Which version do you recommend?

lordofhyphens, 2018-05-07:
@Annaf12 I have no idea; as I said I don't use VS at all. @bubnikv was the original author for the instructions and can provide better guidance.

N4rki, 2018-05-08:
I don't see any way to build Slic3r with Visual Studio at this point either. I spent a lot of time trying to get it to work but had no success. bubnikv and another dev actually made a precompiled perl version according to his old manual and with quite some additional improvements over at Slic3r Prusa Edition but there seems to be no way to use it for the upstream Slic3r. 

Maybe it's a good idea to put some sort of disclaimer at the start of the guide saying that nobody has been able to compile it with VS for a while now? 

Adding to what @Oekn5w said: Another way to code for Slic3r is using Eclipse Cpp + EPIC plugin. After importing the complete codebase as a project it also has realtime error checking for C++ (probably not quite as good as VS2017 though) and through the plugin it does the same for Perl code. You can even run it from the IDE and debug all Perl code using breakpoints etc. You just can't step into any C++ code or set breakpoints. I just used CLI printouts to know what happens inside. Apart from that the compiler lets me know of most of my errors.
Of course this also means that if you changed something in C++ and you want to see what it did, you'll have to compile via MinGW again, which takes about 10 minutes.

lordofhyphens, 2018-05-08:
@N4rki with the current build system you can actually configure local::lib to point to the right spot and so long as certain headers don't change (usually libslic3r.h), you can just rebuild using xs/Build and install from there.

lordofhyphens, 2018-05-08:
If people want to help with my port to wxWidgets C++ UI, head over to https://github.com/lordofhyphens/Slic3r/issues and pull down a copy of the cppgui branch.

That builds with CMake and I am doing what I can to make sure things function with (nearly) every commit.

lordofhyphens, 2018-05-08:
That is probably the fastest route to getting a building system with MSVC :D if you really have to have MSVC.

lordofhyphens, 2018-07-09:
Visual Studio 2013 is not supported (Slic3r uses the c++11 feature constexpr, which is not supported by VS2013).

VS2017 is recommended if you are building cppgui, but VS2015 may work. 

https://msdn.microsoft.com/en-us/library/hh567368.aspx

closed 2018-07-09 by lordofhyphens