slic3r/Slic3r #4406: Gyroid infill bug?

scott0122, 2018-05-09:
### Version
1.3.0-dev, Slic3r-master.2018.05.07.1676.8a93fbb.64bit

### Operating system type + version
Windows 10 64bit

### Behavior
My infill setting
[image]
When I place model on this direction
[image]
Infill on some layer seems broken
[image]
[image]

When I place model on this direction, the problem is gone.
[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[Body1.zip](https://github.com/slic3r/Slic3r/files/1985977/Body1.zip)

foreachthing, 2018-05-09:
@scott0122 I think you need to go over here: https://github.com/prusa3d/Slic3r/issues/

Also, see here: https://github.com/prusa3d/Slic3r/issues/824
And maybe this one, too: https://github.com/prusa3d/Slic3r/issues/825

bubnikv, 2018-05-09:
> Also, see here: prusa3d#824 <https://github.com/prusa3d/Slic3r/issues/824>
> And maybe this one, too: prusa3d#825

Yes, it is  prusa3d#825 <https://github.com/prusa3d/Slic3r/issues/825> and
there is the fix included.

foreachthing, 2018-05-09:
@bubnikv would you be kind enough to PR your changes to Slic3r as well? Thanks!

supermerill, 2018-05-09:
I'm not at home this week. I can't test a slic3r build right now.
I should have PR the bubnikv changes, but as it's a bit time consuming, i waited the next release of slic3rPE. As it's now out, I promise to test it & PR as soon as possible. If someone want to Merge back the changes & fix before me, no problem.

lordofhyphens, 2018-05-22:
@foreachthing that should be a separate Slic3r issue then.

KillyMXI, 2021-07-24:
I experience this issue with gyroid pattern in Slic3r 1.3.0.
Unfortunately, with some of my models, 90deg rotation just makes the issue appear in a different part of a model.

Yenya, 2023-06-15:
I see the same problem with this part:

https://github.com/gregsaun/bear_extruder_and_x_axis/blob/master/printed_parts/stl/mk3s/x_carriage.stl

0.2mm layers, 0.4mm nozzle, 4 solid top and bottom layers, 20 % gyroid infill, min 4 perimeters. After the last solid bottom layer there is a single layer with complete gyroid infill, but the layers above it up to about 2.4 mm have almost half of the infill missing.  Honeycomb works as expected.
[image]