slic3r/Slic3r #4407: Invalid mid-air supports!

Coder-256, 2018-05-10:
### Version
Slic3r 1.3.0 (finally! 3ba8e0f)

### Operating system type + version
macOS High Sierra 10.13.4

### Behavior
* **Issue**: Supports are generated in mid-air in my model. This makes the exported GCODE completely unprintable. ☹️
* **Steps**:
  1. Load the included config and STL.
  2. Rotate 90 degrees around the X axis. (sorry)
  3. Look at the `Preview` and `Layers` tabs.
* **Expected Result**: The bottom of the supports would touch the model.
* **Actual Result**: The supports are floating. (also, why do all supports have different infill?)
[image]

#### STL/Config (.ZIP) where problem occurs
[Slic3rBugReport.zip](https://github.com/slic3r/Slic3r/files/1989947/Slic3rBugReport.zip) (mind my lunatic design)

lordofhyphens, 2018-05-10:
One thing you can do to avoid the problem is edit lib/Slic3r/Print/SupportMaterial.pm and tweak PILLAR_SIZE and PILLAR_SPACING (lines 26 and 27).

It looks like the defaults don't play nice on this model.

You don't need to recompile to see the changes, just close and reopen Slic3r.

Coder-256, 2018-05-10:
What exactly do they do, and what should I set them to? Why is there no config option for this?

lordofhyphens, 2018-05-10:
@Coder-256 It's really easy to get weird results. 

"PILLAR_SIZE" is the X/Y dimensions of the support pillars generated, PILLAR_SPACING is how close  those pillars are allowed to be to each other. 

With your config, 5/15 seems to avoid the problem (which is probably from the pillars because the sloping area makes them too thin).

Coder-256, 2018-05-10:
You're right, that works. Thanks! I guess one possible way to fix this bug is to add those options to the config?

lordofhyphens, 2018-05-10:
I'd probably accept a pull request to add them, but the UI would need to hide the options when it's not pillars type. 

I am a bit busy to go in and add it myself (working on an UI overhaul).

Coder-256, 2018-05-13:
This is mostly fixed by #4410, so I'll leave it up to @lordofhyphens whether this issue should be closed, or if further steps should be taken (maybe implementing some kind of detection/warning for invalid pillar supports).