slic3r/Slic3r #4408: odd line on z-axis

bill-orange, 2018-05-11:
### Version
1.2.9

### Operating system type + version
Windows 10 April 2018 release

### Behavior
* I am seeing an odd line going up the z-axis on one of the demo files furnished with my Anet A8.  When generating code with Cura (which has other problem) the line is not there.  It does not appear to be part of the model.  The line is 180 degrees from the z-axis origin

* _Steps needed to reproduce the problem_
  generate g-conde from the Anet files Chess.stl and print on Anet A8

* _Expected Results_
No line

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
Done

This is likely to be a blunder by me, but since it seems not to occur in Cura, I thought it best to let the experts have a look.
[config.zip](https://github.com/slic3r/Slic3r/files/1993803/config.zip)
[image]

lordofhyphens, 2018-05-11:
2 things: 
Please try using 1.3.0 or 1.3.1-dev
https://github.com/slic3r/Slic3r/releases/tag/1.3.0

Dl.slic3r.org/dev
Also please post a screenshot of Slic3r's preview of the area.

I suspect this might be a seam, but it doesn't quite look like one.

bill-orange, 2018-05-11:
Yea, it looks a little deep and wide to be a seam.  It is also odd that it stops before it gets to the top.  I will reprint the the daily build later.  
[image]
[image]

bill-orange, 2018-05-11:
Using version 1.3, the line is slightly less noticeable using the same setting.

gtremb, 2018-05-11:
I noticed also the "line" using 1.3.0 dev but it was during the printing that I saw it. On my side, it seems to be linked to the "Seam Position". When I select "Random", it is not random anymore on many parts that I printed. When printing a cylindrical shape, there is 3 or 4 lines for example.