slic3r/Slic3r #4409: Infill not joining internal perimeters

arrickg, 2018-05-11:
-Slic3r Version 1.2.9
-Windows 64-bit OS, x64-based processor
-Similar issue(s): #1144 -- "Option for improving perimeter + infill sticking together" 

I am having problems with infill joining the internal perimeters. I have tried increasing the overlap percentage, changing the nozzle size and settings, and changing temps. My current parameters work well for the external perimeters. However, the smaller, internal perimeters remain unconnected. This is problematic as I often print small parts with many cavities and holes.

I suspect that this has something to do with the accuracy limitations and mechanical backlash of my machine (a Hyrel Hydra 640); however, having additional flexibility with the gcode generation would be helpful. Would it be possible to create two distinct overlap percentage settings -- one for the internal perimeters, and one for the externals?

I am currently using a .30mm nozzle with ABS. Head temp is 250 degC, bed is 125 degC, heated chamber at 60 degC. Speeds are all 10mm/s, and I am using two perimeters with 100% infill (perimeters before infill). For more details, please refer to the attached a copy of my config as well as the sample STL. The subject holes are clearance holes for a 4-40 screw.

[Arrick_Slic3r_Issue.zip](https://github.com/slic3r/Slic3r/files/1995635/Arrick_Slic3r_Issue.zip)

Below is a  screen shot of a layer from Slic3r, as well as a picture of the print in progress...

[image]

[image]

lordofhyphens, 2018-05-11:
Try either 1.3.0 or 1.3.1-dev.

A 3d preview screenshot is better data for this kind of thing than the 2d one.

Tinchus2009, 2018-05-14:
can this be related to #4396?

arrickg, 2018-05-14:
Hello @lordofhyphens, 1.3.0 does not seem to offer infill overlap settings for external and internal perimeters respectively, and the problem persists with infill not joining.

@Tinchus2009, the slicing and g-code generation appears to be correct (slightly different than #4396)

I have increased the flow parameter on my printer (a setting external to Slic3r and in my), and this seems to have helped. Having said that, It would still be nice to have the feature I suggested above.

Any other suggestions?

Thank you all for your help so far.

lordofhyphens, 2018-05-14:
Could you clarify what you mean by "external" perimeters? 

For me it means the perimeter furthest away from the local model area.

foreachthing, 2018-05-15:
> I have tried increasing the overlap percentage...

Note1: some infill-patterns do not respect that setting!!
Note2: trying different overlap settings  ...  **none** seem to do anything anymore in 1.3.1.!

arrickg, 2018-05-22:
@mrvn Thank you for your reply. Yes, I think that what you described is what's happening in my setup. As I noted above, the deviation between my designed and measured spec is =< 20%. I have tried using PLA, and is has the same problem. My chamber is heated to 60 degC

I believe that a fix could be a software feature that detects small perimeters and allows the user to bump the overlap for those specific perimeters. Not sure how easy this would be to implement.

mrvn, 2018-05-28:
I don't think the overlap is the problem. The shrinkage is. Imagine your inner hole 10-20% bigger and everything would be fine. A fix would be to model the amount of distortion you get from tension while printing and shrinkage from cooling. And I'm not sure this wouldn't be highly filament and temperature specific.

For example what if you raise your temperature 5° so the extruded filament is more liquid and isn't pulled inward as much when printing small holes? Or go faster or slower? How much does the distortion change with that?