slic3r/Slic3r #4412: Avoid crossing perimeters is not working

Tinchus2009, 2018-05-14:
### Version
1.3.1-dev (laterst win64 compile)

### Operating system type + version
win10 64b

### Behavior
With some material, specially PETs filaments, stringing is something you will have and very dificult to avoid at all. For printing some models, it is very usefull to use the "avoid crossing perimters" options, so you dont have stringing inside that model. 
It seems that option is not working now?

This is a screenshot showing the paths (red lines are travelling moves). As you can see, the path is indeeed crossing perimeter (Im just posting a screenshot of a single layer, but the crossing happens dureing all the printing). I have atached the config, and generated gcode
[image]

[pruebas_impresion.zip](https://github.com/slic3r/Slic3r/files/2002971/pruebas_impresion.zip)

lordofhyphens, 2018-05-14:
The configuration is "avoid", not "never", so it's not going to try forever or do complicated moved to avoid it.

Tinchus2009, 2018-05-15:
ok, I understand. But looking at the movements, do you think that the optional movement to not cross those perimeter would have been complicated ones? In the specific screenshot, the alternative movement would have to move the noozle just 2 mm t one side.

lordofhyphens, 2018-05-15:
@Tinchus2009 yeah, but Slic3r doesn't necessarily know that. Study MotionPlanner.cpp and see if you can come up with a better pathfinding algorithm; shortest_path() is how it's calculated, each island is some place to be avoided (in general).

mrvn, 2018-05-22:
What red line are you talking about? You circled a number of them. I first assume the one going /\ over the blue inner perimeter. But isn't that actually 2 moves, one arriving, then the blue inner perimeter, then leaving?

The only other possibility seems to be the two lines going left way off screen. What's there that it needs to move there? Unless there is nothing there then I don't see a way to avoid crossing perimeters to get there.

foreachthing, 2018-05-22:
> What red line are you talking about?

My guess is: e-v-e-r-y single one inside this hole! If I have a critical part (where I need a clean inner diameter) I use Cura. Sucks, but that's the only way to avoid stringy islands - so far.

closed 2018-05-31 by Tinchus2009