slic3r/Slic3r #4413: Crash using adaptive layer height AND relative first layer height

ilcarogna, 2018-05-15:
### Version
1.3.1.dev

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Slic3r-master.2018.05.13.1690.4fa6251.64bit

(sorry - downloaded directly from the download page . hope this is enough )
### Operating system type + version

Windows 10 64bit

### Behavior
Program freezes and computer freezes when I select adaptive slicing AND relative first layer height (i.e. put  150% in it). Hard reset needed and reboot.

* _Describe the problem_
After loading a simple STL model I started playing with the new setting. I selected adaptive slicing and relative first layer height.

* _Expected Results: Since I am not defining any absolute layer height I did expect an error message or some forced default value for the first layer height (I was wondering how the adaptive slicing was supposed to work without any absolute or default layer height value),

* _Actual Results: Program freezes and computer became irresponsive, forcing a hard shutdown,
  
No screenshot, of course!

Hope this helps. I was actually expecting some form of warning or error.

[test.zip](https://github.com/slic3r/Slic3r/files/2006634/test.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2018-05-15:
Any thoughts @platsch ?

This probably should trigger an error condition in the validation.

lordofhyphens, 2018-05-15:
@ilcarogna FYI the last string before the ".64bit" is the git commit, so you can paste that in and it'll get sorted out.

lordofhyphens, 2018-05-15:
@ilcarogna 
Could you add the following lines in the validate sub in lib/Slic3r/Config.pm, restart Slic3r and see if it errors out appropriately?
````
    die "Adaptive slicing requires a non-relative first layer height."
        if $self->get_value('adaptive_slicing') && $self->first_layer_height ~ /^(?:\d*(?:\.\d+)?)%$/;
````

It looks like it should function according to my online regex tester (and applied it to the build, you adding it to that sub would just mean it is quicker to test for you).

lordofhyphens, 2018-05-16:
Fixed via adding a configuration validation check to prompt an error on attempt to slice.

closed 2018-05-16 by lordofhyphens

platsch, 2018-05-16:
Didn't track the problem down yet, but I'm almost certain that it is not directly related to adaptive slicing. I can reproduce it with static layer heights too. And it didn't occur a few weeks ago.

I run into the problem when background processing is enabled, and I start typing the relative value e.g: 120%. Background processing kicks in after each digit and the slicing process is triggered with 1, 12 and 120mm which leads to a crash. C&P 120% into the field works as intended.

bubnikv, 2018-05-16:
You will get a lot of similar issue until you will get the multi threading
under control. That means to get rid of the perl threads, to review
possible data access collisions and introduce mutexes where needed.

I have already started that work in the Prusa3D fork, but it is quite a lot
of work and I did not manage it in 1.40. I hope to finish that in 1.41. It
helps a lot, that we have now one dedicated tester in our team.