slic3r/Slic3r #4414: Crash on loading Layers tab after changing support material

TheThirdOne, 2018-05-16:
### Version
This was tested on 2d064e2569b0c.

### Operating system type + version
This is on Linux Mint 18.2

### Behavior

I stumbled upon a crash while testing. 

Steps to Reproduce:
  - load a model
  - go to the preview tab
  - wait for the preview to fully generate
  - toggle generation of supports
  - before the preview is generated, go to the Layers tab

I have gotten a few different logs from doing this (on different runs).
Run with the perl debugger:
```
Range iterator outside integer range at pathToSlic3r/lib/Slic3r/Layer.pm line 31.
```

Run in any way:
```
Out of memory!
*** Error in `/usr/bin/perl': free(): invalid size: 0x00007fffbcab1500 ***
======= Backtrace: =========
...
Nothing useful.
```

Or run normally:
```
Segmentation fault
```

#### STL/Config (.ZIP) where problem occurs
Happens with a default config and this model I have been using for testing [D20.zip](https://github.com/slic3r/Slic3r/files/2007444/D20.zip)

TheThirdOne, 2018-05-16:
A simple workaround to this is to simply wait before switching.

lordofhyphens, 2018-05-16:
It probably has something to do with the preview not being available yet when the layers tab attempts to load it.