slic3r/Slic3r #4415: Generated GCode has too much gaps between soild layers and perimeters

makkarpov, 2018-05-16:
### Version
1.3.0-dev, commit c206ae7
### Operating system type + version
Linux, Fedora 26, `Linux <hostname> 4.12.9-300.fc26.x86_64 #1 SMP Fri Aug 25 13:09:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

### Behavior

Slic3r generates too sparse perimeters and solid layers, which results in a poor print quality. I'm not sure whether it is a slic3r bug, or I had missed something in configuration, but I had looked through all options and none of them seems to be related to this. 

Consider following simple test cube:
[image]

Cura produces following toolpath on a layers with soild infill:
[image]
As you can see, infill definitely overlaps with perimeters and slightly overlaps with itself, resulting in good quality print with no gaps. Perimeters also slightly overlap with each other.

Slicr3r produces following toolpath:
[image]
As you can see, infill just barely touches perimeters, and has a visible gaps. Moreover, perimeters are also too far from each other. This results in a severe degraded quality of print.

#### STL/Config (.ZIP) where problem occurs

* Model itself: https://www.thingiverse.com/thing:1278865
* Slic3r config used: https://files.makkarpov.ru/tmp/slic3r-perimeters-issue/slic3r-cube-config.ini
* Cura profile used: https://files.makkarpov.ru/tmp/slic3r-perimeters-issue/cura-cube-config.ini

lordofhyphens, 2018-05-16:
Increase infill overlap %. The default is ~60%, your config has it at 15%.

closed 2018-05-16 by makkarpov

lordofhyphens, 2018-05-16:
You can also set it to a flat mm value, just FYI. I hope you are getting good results out of Slic3r now :)

lordofhyphens, 2018-05-16:
You should also confirm layer toolpaths with Slic3r's own 3D Preview. Most gcode view tools don't know the line widths to use and are as such inaccurate. Slic3r's 3D Preview does know the width that the slicer thinks it is using.