slic3r/Slic3r #4417: Modifiers creating disjoint object

jmgiacalone, 2018-05-16:
### Version
Git master commit 4fa6251dd5a...

### Operating system type + version
Ubuntu 18.04 64-bit

### Behavior
Using a modifier mesh creates extra unwanted perimeters, leading to disjoint parts being printed.
[image]
[image]

The modifier is created as a lambda-box and has extra infill density set. I found some old closed issues, #2381 and #2690, where @alexrj states the extra perimeters should be generated if the perimeter settings are altered by the modifier mesh, but adjusting the infill density does not fall into this category.
What I would actually like to implement is a slow down of the perimeter speed when printing overhangs. I am also investigating a way to do this automatically, but in the mean time I thought I could produce a work around using modifier meshes. But adjusting the perimeter speed using a modifier also generates extra perimeters.

In fact, the Slic3r manual page explaining modifiers suggests this should work without creating extra perimeters, although it is not clear from the image whether the extra perimeters are indeed created, but if they were, this would result in a weak joint in the part which would make the 100% infill density rather useless.

lordofhyphens, 2018-05-16:
Please upload an exported AMF of the configuration and a config export.

jmgiacalone, 2018-05-16:
It's not a complex part, but here they are...

[10mm-cube.zip](https://github.com/slic3r/Slic3r/files/2010961/10mm-cube.zip)

lordofhyphens, 2018-05-16:
[image]

jmgiacalone, 2018-05-17:
I am sorry, I should have spotted that. Thank you for taking the time to figure this out.

foreachthing, 2018-05-17:
@jmgiacalone: if you don't want to "disturb" the bottom and topmost layer, just scale your modifier to be `h_modifier = h_model - 2 * <layerheight>`and move in +Z one layer height.

lordofhyphens, 2018-05-17:
@foreachthing the way that xy compensation currently works it operates per region. So it didn't matter if the modifier mesh did not cross the top surfsce.

lordofhyphens, 2018-05-17:
@jmgiacalone that is why we require a test model and config, no matter how simple it looks :)

foreachthing, 2018-05-17:
@lordofhyphens what I meant was this: to get rid of those "artifacts" 
[image]

scale the modifier to be (at least) two layers shorter in Z:
[image]

Result:
[image]