slic3r/Slic3r #4420: Crash on Undo

TheThirdOne, 2018-05-17:
### Version
This was tested on ae7fdfe227.

### Operating system type + version
This is on Linux Mint 18.2

### Behavior

I stumbled upon a crash while testing. Its kindof intermittent; it works considerably better with the small d20 I have been using than the larger one.  

Steps to Reproduce (for most consistent results, do it quickly):
  - load a model
  - click off the model
  - hit undo twice

I have gotten a few different logs from doing this (on different runs).
Run with the perl debugger:
```
Signal SEGV at pathToSlic3r/lib/Slic3r/GUI/Plater.pm line 2587 thread 1.
	Slic3r::GUI::Plater::__ANON__[pathToSlic3r/lib/Slic3r/GUI/Plater.pm:2596]() called at pathToSlic3r/lib/Slic3r/GUI/Plater.pm line 2600 thread 1
	Slic3r::GUI::Plater::__ANON__[pathToSlic3r/lib/Slic3r/GUI/Plater.pm:2600]() called at pathToSlic3r/lib/Slic3r/GUI/Plater.pm line 2600 thread 1
	eval {...} called at pathToSlic3r/lib/Slic3r/GUI/Plater.pm line 2600 thread 1

```

And when run normally most of the time:
```
Segmentation fault
```

But one time I got:
```
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadIDChoice (invalid resource ID chosen for this connection)'.
  (Details: serial 18319 error_code 14 request_code 53 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

#### STL/Config (.ZIP) where problem occurs
Happens with a default config and this model I have been using for testing 
[d20s.zip](https://github.com/slic3r/Slic3r/files/2014987/d20s.zip)

TheThirdOne, 2018-05-17:
@Samir55, interesting. I wasn't aware it was already known. 

I did a bit of looking around on this one and it appears to be caused by a race condition on thumbnails.

TheThirdOne, 2018-05-24:
Ok, I've done some more investigating on this. 

The crash appears to only happen if [make_thumbnail](https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm#L2580) is called at least 3 times in quick succession. I don't know why it appears to be that 3 rather than 2 triggers it, but my code that calls it twice pretty much never activates the bug, whereas my code with 3 triggers it everytime.

I also don't know what the actual bug is. However, I do believe that I have a hacky fix. Simply putting a guard inside the function so that if a thread is already generating a thumbnail, then it will just return immediately seems to fix it.

I am going to do some more testing and move my fix over to the current master to see if I can validate that it is indeed preventing a crash.

lordofhyphens, 2018-05-24:
@thethirdone it is probably Perl threads; something is probably managing to access the memory location that is out of bounds.

TheThirdOne, 2018-05-24:
So, I think I have narrowed down the problem to two cases:
  1. If `$plater_object` is valid when starting the thumbnail but not when calling `$plater_object->make_thumbnail`
  2. make_thumbnail is called multiple times in quick succession causing `$plater_object->thumbnail` to be invalid before the call to `$plater_object->make_thumbnail`

Solving the second is relatively simple to work around as I laid out in my earlier comment, but the first is harder to solve. However, most of the crashes I have experienced have been due to the second issue; it is a lot easier to do actions in quick succession compared to something which only occasionally happens on undoing adding objects.