slic3r/Slic3r #4421: Feature request: Support Material settings on modifier session / selective support generation

thiagopeixoto16, 2018-05-18:
In order to enable a **selective support generation**, my suggestion is to use the already implemented Modifier logic.  
It is intuitive and I think relative simple to implement. 
Basically add all support settings on Modifier Part Settings.

For instance, the object below needs support only on that circled part. I tried (instinctively) to apply a support material on the modifier but it is not available there, so I think that solution is intuitive.
[image]
In that case unfortunately I had to use meshmixer to generate the support only on that part.

Please let me know if it is a good solution. 
I didn't find any related issue proposing that, please merge this issue if I failed to find.

Let me know if more information is required.

lordofhyphens, 2018-05-18:
I have been considering this for quite some time. The primary barrier to doing so is that there is currently no information about the modifier meshes at the stage the support material is generated.

thiagopeixoto16, 2019-02-18:
I've noted that Prusa has implemented my suggestion:
https://www.youtube.com/watch?v=CSqX_8pVi2o
Is there any way to pull this from PE version?

lordofhyphens, 2019-02-18:
@thiagopeixoto16 you would have to port their entire support system. And provide the unit tests that I will require to make it a PR. If Prusa's devs have written any unit tests for their code, then I haven't been able to find them. 

Since writing the appropriate tests involves having an intimate understanding of how the supports are supposed to work...