slic3r/Slic3r #4428: Adding and/or removing instances via keyboard does nothing

foreachthing, 2018-05-24:
### Version
1.3.1
Slic3r-master.2018.05.23.1703.f3b5909.64bit

### Operating system type + version
Windows 7 x64, English, Swiss keyboard

### Behavior
* [CTRL]+[+] _does not_ add another instance of model.
* [CTRL]+[-] _does not_ remove instance.
* Note: [CTRL]+[Delete] _does_ remove model.

### Expected behavior
* adding or removing instances.

#### STL/Config (.ZIP) where problem occurs
ANY model will do, but I know somebody will ask for one...
so, here's one: https://github.com/alexrj/Slic3r/files/1614215/cube10x10x3.zip

xoan, 2018-05-24:
You need to select the object in 3D or 2D view to use the keyboard shortcuts.

PS. It works as expected on my system (Arch Linux + Xfce + Slic3r current master)

foreachthing, 2018-05-24:
@xoan, ya, well duh! :-)  But, I'm on Windows ...

xoan, 2018-05-24:
What «+» and «-» keys are you using, keyboard ones or numpad ones?

foreachthing, 2018-05-24:
the ones on the numpad. Otherwise it would look like this: [CTR]+[SHIFT]+[1] (+ is on top of 1 on the Swiss keyboard). Which does not work either - just tried it.

I tried Ubunut (VirtualBox) with the onscreen (and normal) keyboard and it does not work either.

lordofhyphens, 2018-05-24:
It is probably an issue with the accelerator (or possible translation of).

xoan, 2018-05-24:
In Linux, using keys from Numpad doesn't work. Only work when using «+» and «-» from keyboard.

[image]

lordofhyphens, 2018-05-24:
A fix probably can be done in the key event handler in Plater.pm

foreachthing, 2018-05-24:
Well, how about that!!
[CTRL]+[1] (where the + prints, with [SHIFT]) does actually work!! So does [CTRL]+[-] (dash)!
[image]

xoan, 2018-05-24:
Not dash, but minus.

foreachthing, 2018-05-24:
Unless one can change that to the minus on numpad, one can close here. Thanks!

closed 2018-05-25 by foreachthing