slic3r/Slic3r #4432: Slicer crashes when exporting composition .gcode when slicing certain .STL files.

veive, 2018-05-28:
### Version
1.2.9

### Operating system type + version
Windows 10, latest patch

### Behavior
* _Describe the problem_
Go to https://www.thingiverse.com/thing:74041/#files download dungeon_tileset.stl and slice.
Expected Results A new file named composition.gcode that my printer can use.
Actual Results Slicer freezes, I have tried letting it process for an hour and it did not resolve the issue.

_Is this a new feature request?_
No
[config.zip](https://github.com/slic3r/Slic3r/files/2043217/config.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Samir55, 2018-05-28:
Hello @veive, Please try to use the latest release and see if this issue is still found. The latest builds can be downloaded here https://dl.slic3r.org/dev/

veive, 2018-05-28:
I just tried with the version from Slic3r-master.2018.05.23.1703.f3b595.32bit.zip

It has been frozen with the status bar at the bottom reading 'processing triangulated mesh' for about 10 minutes.

I have also tried with Cura and confirmed that the .stl file is good.

veive, 2018-05-28:
Ok, it took a few minutes but it looks like it did slice with the latest dev release.

foreachthing, 2018-05-28:
Try **[this ](https://github.com/slic3r/Slic3r/files/2043608/Blender3DPrinting-Dungeon_Tileset_Netfabbed.zip)** STL once, please.
I ran it through netfabb and Blenders' 3D-Printing "Make Manifold" and removed double vertices.
It sliced - it took a looooooooooong time to export the GCode! While waiting, I sliced this file with Cura, then CraftWare and even checked my mail... and in the end I was left with the `.tmp` file. Slic3r did not rename it it `.gcode` (but most likely because I opened the `.tmp` file while Slic3r wrote to it).

Attached file: [Blender3DPrinting-Dungeon_Tileset_Netfabbed.zip](https://github.com/slic3r/Slic3r/files/2043608/Blender3DPrinting-Dungeon_Tileset_Netfabbed.zip)