slic3r/Slic3r #4434: imposible extrusions width

Tinchus2009, 2018-06-01:
### Version
all 1.3.1 -dev

### Operating system type + version
win10 64b

### Behavior
This is a behaviour I have been observeing since some time but I think it  is getting worse. Sometimes is is posible to fix it manually setting the extrusion widht of perimeters. But sometimes it is just imposible to get rid of it. Problem is that there is no way a 0.4 extruder (in my example) can extrude such extrusion like the ones you can see in the images. And this introduces some problems to the prints: 
1.- For some material prone to oozing, makes than when this extrusion are meant to be done... a lot of blobs are actually extruded because of all the oozing.
2.- A lot of extra unnecesary printing time is added.
3.- Since actually almost no material is extruded, the pass od the noozle trying to create them , just heat up the printed area... Also, for some materials like PET, if the layer is big, lets say 20 seconds to print that layer, we are talking that the filament is staying in the melting chanber that time, causing a OT of oozing, and risk of jamming because of hight temperatures.

There should be a limit to the minimun extrusion width. For a 0.4 noozle, extrudeing  something smaller than 0.2 is almost imposible in real terms

A separate question: there is a script in the utils folder, that is suposed to removed small extrusions. How does it works, o, it works?

Example image

[image]

Another example

[image]

lordofhyphens, 2018-06-01:
That's gap fill. Turn it off if you don't need it.

Tinchus2009, 2018-06-01:
Yes, that is what I do when I can not use gap fill. Thing is that I need gapfill... In the above example, I have like 50 layers with that "defect" and then 200 layers or more with "good"gapfill.

lordofhyphens, 2018-06-01:
@Tinchus2009 Turn it off with a modifier mesh then? 

[decimate.pl](https://github.com/slic3r/Slic3r/blob/master/utils/post-processing/decimate.pl) is a post-processing script. You use it like you would any other (add it to the config string in Print Settings -> Advance). Edit the script itself to specify how tiny of an extrusion you want to remove.

Tinchus2009, 2018-06-01:
mmm yes, a modifier mesh could do the job, but it is like we are trying to use a patch to solve a huge bleeding... What I posted above was just 2 screenshots to show the problem... in a normal print you can find 100, 200 300 layers with this problem, and the layers are not all together. 
If ou look at the root problem you will see that there is a lot of callateral damage made for these extrusions:
If you look at my example images that is a wall made of 6 perimeters. Tehn you waste a 7th pass for makeing that gap fill... that is a waste of 10% of print time in that layer, asssuming the gap fill speed is the same or faster as perimeters. If you consider that the gap speed usually is slower.... And the heat radiated to the other estruded paths?
I report this case to to get a solution for myself but to help improve the software. I think this is one of those cases where a problem makes a cascade of other problems.

Regarding the decimate.pl script: any tip about how to make it work in windows? I can see 2 values:
my $mindist = 0.33;

my $mindistz = 0.005;

What they mean and in the example images wich values should I use?

lordofhyphens, 2018-06-01:
@Tinchus2009 You could start with the defaults. I don't use the script myself. 

($mindist ^ 2) is the minimum length of move to allow through in XY, $mindistz is the minimum Z length move.

Tinchus2009, 2018-06-01:
mmmm so the script would not help here, since it only act on lenght of extrusions

lordofhyphens, 2018-06-01:
@Tinchus2009 add a reproduction model + config, please. 

@TheThirdOne do you want to put some thought into something like this? Areas of the SW of interest would be PerimeterGenerator.cpp around lines 266 and 160-168.  You can probably start by playing around with `min`.

TheThirdOne, 2018-06-01:
Sure, I'll take a look into addressing it.

Tinchus2009, 2018-06-01:
Sure. I have atacjed a model where this "error" is clearly happening all around the wall of the object, and the config. The config has all widht setting in auto. I could get around this modifying the width values and useing manua ones, but that would work ONLY in this particular model because it is simple. More complex models... well, it would be really alm ost inposible to do that.

[F.zip](https://github.com/slic3r/Slic3r/files/2063964/F.zip)

Im not a coder but an idea: what is there is some post script that is able to remove any extrusion that is noozle size - some reasonable margin width? For example, if noozle is 0.4, anything less than an extrusion of lets say 0.2 or 0.3 width should be removed?

lordofhyphens, 2018-06-01:
@Tinchus2009 That information isn't available from the GCode itself.

TheThirdOne, 2018-06-04:
I haven't made much progress into understanding this bug. 

It appears the magic number 10 in the code is not super special. Changing it to 1000 and 0 doesn't seem to have any effect on the strange output for the given model. 

The output is different and doesn't appear to have any small extrusion widths if detect thin walls is disabled. So that may be a decent workaround.

lordofhyphens, 2018-06-04:
The value is probably a scaled number,  so try 10k or 100k

Tinchus2009, 2018-06-27:
May be the easiest thing could be a post processing script to removed all extrusions shorter than X and thinner than Y? and haveing the ability to configure X and Y as variables so the user can chose what to remove?

lordofhyphens, 2018-07-09:
@Tinchus2009 can you check again with cc42bb411c1239bdfd2efc89bd9126f3f8f1d161 or newer? There were some changes around thin walls.

Tinchus2009, 2018-07-11:
Sure. Done: still is there:

[image]

Also, thanks to this test I have just discovered the source of a problem I have been haveing, I will create a new post about it.
Test about this issue here were done useing objectand config already posted here.

Tinchus2009, 2018-07-11:
I have also tried to disable gap fill. Situation improves a lot, but still those extrusion appear in parts of the model:
(this is the same model: if we see that extrusion, the defaul is 0.44 so that extrusion in the image should be like 0.1 or something like that)

[image]

lordofhyphens, 2018-07-11:
@Tinchus2009 You should *not* be seeing those if gap fill is off.

Tinchus2009, 2018-07-11:
No, that image is WITH gpfill on. But disableing gapfill produces a lot of extrusion in the print, unnecesary if we have gap fill. The point is: we must get rid of all the befits of gapfill because of these weird extrusions? The maths of extrusion says that anything below noozle diamter is not good to be extruded. we can take that to some liits, like extrudeing 0.3 with a 0.4 noozle,  but anything below that will just bee to thin to be something really good. The problem with those tiny extrusions: imagine a circle, the extruder goes all around that perimter trying to extrude those thin extrusions. Usieng lot of filaments that are prone to leaking, like PETg, will only result in a lot of blob around the perimeter, and what is worst, at the end of the path, the extruder will be with no pressure, creating a gap at the beginning of next layer/good extrusion, and even creat a clog. These are not theories, this is what Im observing in prints. A limit to the smaller width an extrusion can have would be a solution, no extrusion width should be smaller than 50% of the noozle size

lordofhyphens, 2018-07-11:
@Tinchus2009 your previous post said:
> I have also tried to disable gap fill. Situation improves a lot, but still those extrusion appear in parts of the model:
(this is the same model: if we see that extrusion, the defaul is 0.44 so that extrusion in the image should be like 0.1 or something like that)

So your report there was confusing.

Tinchus2009, 2018-07-24:
Sorry, but what part was confuseing?
1.- if I disable gap fill... yes, those tiny extrusion desapear, but then I lost gap fill where it is really needed, and that would be like in 99% of prints, because these "imposible extrusions" are present in almost all sliced models.
2.- if I enable gap fill, those extrusions are still present in cc42bb4