slic3r/Slic3r #4436: How to add custom gcode "after tool change"

MJ2020, 2018-06-01:
Hi;

I want to add some custom gcode "after tool change". When I use "tool change gcode" text box, it add gcodes before tool changes, but I need to add them after tool change (I mean after moving X and Y offset). Can you help me?!

lordofhyphens, 2018-06-01:
The toolchange-gcode is inserted immediately before the "Tx" command (command to change tool). 

Is there something odd going on with your setup is doing mechanical actions as part of the toolchange?

MJ2020, 2018-06-01:
thanks for your reply.
Yes, in fact there is something odd :))
I want to move Z up before tool change and then move Z down after tool change. So I need to add some gcode after moving offsets!

MJ2020, 2018-06-01:
Hi;
I did it as a post processing script. It's my first Perl code! the only thing is that I added a comment (;ToolChange) in Toolchange custom gcode in slic3r.

`#!/usr/bin/perl -i                                                     
use strict;
use warnings;

if($^O=~/^MSWin/){
	$^I = '.bak';
}

my $Z_UP=0;
while (my $thisLine=<>) {
	if ($thisLine=~/^;ToolChange/){
		$Z_UP=1;
		my $MoveUp="\nG91\nG1 Z20 F1000\nG90\n";
		print $thisLine.$MoveUp;
	}elsif($thisLine=~/^G[01](\h+X(-?\d*\.?\d+))?(\h+Y(-?\d*\.?\d+))?(\h+Z(-?\d*\.?\d+))?(\h+E(-?\d*\.?\d+))?(\h+F(\d*\.?\d+))?(\h*;\h*([\h\w_-]*)\h*)?/){
		# G0 and G1 moves
		my $X=$2, my $Y=$4,	my $Z=$6, my $E=$8,	my $F=$10, my $verbose=$12;
		# only X&Y movement
		if(!($E) && !($Z) && ($X || $Y) && $Z_UP){
			$Z_UP=0;
			my $MoveDown="\nG91\nG1 Z-20 F1000\nG90\n";
			print $thisLine.$MoveDown;
		}else{
			print $thisLine;
		}
	}else{
		print $thisLine;
	}
}`

lordofhyphens, 2018-06-02:
Not a bad way of solving your problem. ;)

nwneisen, 2019-09-02:
Is there a desire to add an `after-toolchange-gcode` command line option?  Something similar to `end_gcode` and others.

lesikiel, 2021-09-21:
Although it is quite an older thread, I would be highly interested to insert an after-toolchange-gcode command line option. That way we can use conditional gcode to do some tool specific tasks AFTER the tool is changed! It would be highly appreciated.