slic3r/Slic3r #4438: Rotating Face to Bottom: Scale, after scaling and rotating, is set back to 100%

foreachthing, 2018-06-05:
### Version
Slic3r-master.2018.06.05.1711.ff3234d.64bit

### Operating system type + version
Win 7 x64

### Behavior
* Import and scale a model to something other than 100 % (i.e. 65%)
* RF2B
* Go back to scale and observe! It's now set to 100%.

### Expected Results
* Scale remains at previously set value of 65%.

#### STL/Config (.ZIP) where problem occurs
**_ANY_** model and config will do ... try it with _your_ favorite model - please!

xoan, 2018-06-05:
Confirmed (ff3234d in Arch Linux)

lordofhyphens, 2018-06-05:
@TheThirdOne :)

I'm glad I merged this one early.

TheThirdOne, 2018-06-05:
Is this not the expected behavior? It occurs the same way with other rotations (excluding Z). 
I have confirmed that it occurs the same way for normal rotations in 1.3.0.

Pretty sure this is because the model is being recalculated when rotated (a special case allows this to be left out for z rotations). It should be possible to do something to make this act in the expected way relatively easily, but I am not familiar wit the scaling code at all, so I would have to spend time reading before I could say for sure.

foreachthing, 2018-06-06:
@TheThirdOne, ooooh! I did not test that! *shame*
I am sorry! This has nothing to do with RF2B ... this is "normal" Slic3r behavior - which isn't what I expect either.

Oekn5w, 2018-06-09:
I'm still working on transformation matrices, so that the meshes in the volumes don't get changed at all. This will obviously touch many core functions and classes (instance, object, volume; rotations/ translations/ scaling in plater and settings; undo/ redo), so I will probably come across this when implementing the class.
That said, it will take another month at least (quite busy rn), so if someone wants to make an intermediate fix, feel free, but I'd wager it a low-priority issue.

foreachthing, 2018-06-09:
Low priority, my words! Just found it by accident.

TheThirdOne, 2018-07-20:
Ok, did some looking into this. It turns out there is a simple 3 line fix. Just save the scaling before rotating and set the scaling to 1. Then after rotating reset the scaling. A little hacky, but super simple. 

However, that doesn't address all of the other places that the scale gets reset. Namely the rotate face and cut dialogs.  Interestingly opening either of the dialogs (even if you cancel) will reset the scaling. The rotate face dialog doesn't need to do this; that line can simple be deleted. However, it seems like the cut dialog does need it to function normally. 

If that solution sounds good I can make a pull request.