slic3r/Slic3r #4439: 2D View shows wrong oriented model

foreachthing, 2018-06-06:
### Version
Slic3r-master.2018.06.05.1711.ff3234d.64bit

### Operating system type + version
Win 7 x64

### Behavior
#### 1
* Import [any]-model
* confirm model in 2D-tab (check!)
* back to 3D: rotate model in any other axis than Z
* 2D: model remains the same as before. Bad.
---
#### 2
* 2D: rotate model in Z (45°-button)
* confirmed in 3D! Good!
* 3D: rotate model BACK (in Z -45°)
* 2D confirmed. Good
---
#### 3
* 2D: Rotate model 90° in X
* 2D: model not updated! BAD!
* 3D: model good.
---
#### 4
* 3D: Rotate model 90° in X then 45° in Z.
* 2D shows the original imported model, only rotated 45° in Z ...
---

2D is somehow out of sync. Whatever is in 3D, is in Preview and Layers as well.

Some screenshots from the SAME oriented model. Cycling through the tabs:

[image]

Wooohaaa!
[image]

[image]

[image]

#### STL/Config (.ZIP) where problem occurs
Use your favorite model and fastest slicing profile.

Oekn5w, 2018-06-07:
Cause:

https://github.com/slic3r/Slic3r/blob/1527cfbb26b8494c10a7868f36510aaa1768a43b/lib/Slic3r/GUI/Plater.pm#L2644-L2650

return condition is true except the initial loading of the object, although the variable isn't used anywhere else except the declaration?!
Threading issue?

@TheThirdOne, you seem to have added this a week ago?

TheThirdOne, 2018-06-07:
Ok, this came from my misunderstanding of how perl threading works. `plater_object` gets copied which means that resetting in the other thread does not have an effect. Moving the reset to `on_thumbnail_made` works as intended.

@foreachthing, in case you were wondering, the reason Z rotations don't exhibit the bug is that the thumbnail is not retaken on Z rotations. It can just be rotated.

closed 2018-06-09 by lordofhyphens

foreachthing, 2018-06-13:
Another issue with 2D:

3D: 
[image]

2D:
[image]

Model:
[Gear_Pump.zip](https://github.com/slic3r/Slic3r/files/2098340/Gear_Pump.zip)

lordofhyphens, 2018-06-13:
@foreachthing please open a second issue for the other problem.

PR #4443 (2018-06-09): Moved resetting parameter of thumbnail generation to on_thumbnail_made
Fixes #4439 introduced in #4424.