slic3r/Slic3r #4440: Maximum Filament Width Increase Request

LandSquidConfirmed, 2018-06-06:
Slic3r 1.3.1

Discovered Problem on Windows 10, probably present on MacOS and Linux

For large 3D printers, the maximum extrusion width of 1.7mm is prohibiting. (This is a problem for at least me personally with a 3mm nozzle) Because this seems like an arbitrary limit, I was wondering if there was some way to change it by editing Slic3r files, and hoping that it could be changed by default.

Providing support for extremely large printers would be great in Slic3r, because no other software seems to be able to pull off slicing for large printers without many issues.

lordofhyphens, 2018-06-07:
@LandSquidConfirmed Are you talking about the automatic width calculation or what you set via a config parameter?

lordofhyphens, 2018-06-07:
@LandSquidConfirmed Please include a Slic3r config export (ctrl-E) and a 3D model with which you are seeing the issue.

LandSquidConfirmed, 2018-06-07:
I updated the post, hopefully this will help you reproduce the issue.

lordofhyphens, 2018-06-07:
@LandSquidConfirmed I need an exported config, not screenshots. See the highlighted menu option in the screencap below:
[image]

Reference: https://github.com/slic3r/Slic3r/blob/master/.github/CONTRIBUTING.md

lordofhyphens, 2018-06-07:
@LandSquidConfirmed So the error you are seeing comes from Config.pm around 276. It triggers when you try to have an extrusion width 10x the nozzle width. You can, of course, delete the validation check (doesn't need recompilation).

closed 2018-06-07 by lordofhyphens

LandSquidConfirmed, 2018-06-07:
Thanks buddy mistakes were made.