slic3r/Slic3r #4444: feature request: support for "Octoprint-Cancelobject"

nicksears, 2018-06-11:
I have a very simple, but super simple request to add support for Octoprint's ["cancel object"](https://plugins.octoprint.org/plugins/cancelobject/) plugin.

I absolutely LOVE pronterface's ability to cancel a failing object by selecting it with a box. As I understand it, the program then ignores gcodes within the coordinates of the box. The plugin says it only supports gcode sliced with simplify3D because of the comments made by the program, unique to each object.

This uses a regular expression to look for " ; process (.*)" (excluding quotes) 
Not sure how the "#Object" part works

This would be a REALLY powerful addition to slic3r and any host software using its gcode.
[image]
[image]

lordofhyphens, 2018-06-11:
This is relatively easy to do in a limited fashion, and harder to do comprehensively.

nicksears, 2018-06-12:
I agree, easy/intuitive identification of multiple identical parts would
be pretty difficult... I would love to see the most basic implementation.

Anything I can do to help? I dont think I have the coding skills required
(compiling always gave me trouble)

I didn't originally include it, but the source page might be useful.

https://github.com/paukstelis/OctoPrint-Cancelobject

I've requested he post an example gcode for testing/comparison since I no
longer have simplify3D.

lordofhyphens, 2018-06-12:
Just a FYI: If you are using sequential printing, you could use the between objects custom gcode to insert a marker and then use a post-process script to add the relevant gcode.

paukstelis, 2018-06-14:
Presumably there is some index value that could be appended in the case of object copies? 
It does work with sequential printing, The plugin readme has been updated to reflect this.

paukstelis, 2018-06-25:
I would add that ideally the slicer would also reset extrusion distances between objects. If that didn't happen it could be handled in the plugin, but just requires a bit more overhead.

lordofhyphens, 2018-06-26:
Slic3r issues G92 E0 on every retraction already.

paukstelis, 2018-06-26:
@lordofhyphens, perfect. Is the default behavior to also do G92 E0 when going between different features (infill, supports, outlines, etc.)? Talking with folks at ERRF last weekend, there is some interest in being able to allow cancelling in a more specific fashion, for example, just cancel the supports of an object if you decide you don't need them anymore.

lordofhyphens, 2018-06-26:
Rake a look at the output gcode

supermerill, 2018-07-03:
i made a simple implementation here 8b53b1bbfd009adcbac4d7eae98ab736302070e1

you just have to add some "if (this->config().gcode_comments)" in the perl section (I don't know where this hide in the perl objects) in gcode.pm line 550 & 677 to only activate these comments when "verbose g-code" option is checked.

lordofhyphens, 2018-07-05:
I've merged the PR and changed it to use an option instead of being tacked onto Verbose gcode (verbose gcode adds a lot to file size on its own).

lordofhyphens, 2018-07-05:
@nicksears can you give it a spin and check the output gcode to see if it works for you? The option lives next to verbose gcode in the Print area.

closed 2018-07-05 by lordofhyphens

nicksears, 2018-07-05:
checked the latest build here (https://dl.slic3r.org/dev/win/) but I don't see the option in print settings-output options. I've never had good luck compiling from scratch on my machine, is that the only way, or can anyone point me in the right direction?

lordofhyphens, 2018-07-05:
@nicksears https://dl.slic3r.org/dev/win/Slic3r-master.2018.07.05.1728.75de746.64bit.zip 

It takes a while for the build to get uploaded.

@supermerill I don't actually know what you are talking about.

lordofhyphens, 2018-07-05:
@supermerill nevermind, I figured it out (missed a spot in GCode.pm).

lordofhyphens, 2018-07-05:
@nicksears new build should have 55a4e95b in the name when it is done.

supermerill, 2018-07-05:
@lordofhyphens oups, gcode.pm not .cpp
see 2d222c63851f0dadfcada720785f63fbfff43dee

nicksears, 2018-07-05:
@lordofhyphens 
Gotcha. tried the new one you linked to and it's definitely inserting lines like "; printing object chair locker 061818.STL id:0 copy 0"

Do I need to change the "object regular expression" (default is "; process (.*)" - no quotes) or the "object replacement tab" (default is "#Object" - no quotes) in the CancelObject plugin settings?

I may have an issue with the CanceObject plugin installation altogether, though. It shows up as a selectable tab like temperature/control/terminal, etc, but it's totally blank...

Not sure if it's related, but it gives a cryptography error when I install, something about I would have v 2.1.4 instead of 2.2.1 and the versions being incompatible?  I can't be the only one getting that with a fresh install of everything.

paukstelis, 2018-07-05:
Yes, you need to change the object regular expression to:
`; printing object (.*)`. Make sure you also enable relative extrusion.

nicksears, 2018-07-06:
@paukstelis could you provide your example gcode so I could see if it's my slicing or octoprint/plugin setup?