slic3r/Slic3r #4445: Change Alt to Shift for vertical camera movement.

TheThirdOne, 2018-06-13:
As I was going through the mouse movement code for Scene3D, I noticed there was a [feature](https://github.com/slic3r/Slic3r/blob/ff3234d/lib/Slic3r/GUI/3DScene.pm#L251) to move the camera's target vertically by dragging while holding alt. 

Unfortunately, this isn't something that I can use on my machine because alt-drag is used by the OS to move windows. Changing it to be activated on shift-drag (or ctrl-drag) would make collision with OS level modifiers less likely. 

I can change my configuration to use a different key for moving windows, but until I looked through the code I had no idea that the camera feature existed because I couldn't access it. In case that this isn't a problem unique to my setup, I think it would be good to address it. 

The actual code to change is really simple; so its just a matter of if this should be changed.
Does anyone else have an action bound to alt-drag that overrides this feature? And for people who don't, are you used to it on alt-drag? Any other thoughts?

foreachthing, 2018-06-13:
SolidWorks uses CTRL+Drag ... Blender uses SHIFT+Drag ... so, wth, change it to ALT+Drag.
Three applications, three different controls for the same function.
Another application uses the mouse only: [left click+hold]+Drag.

BTW: I'd like to use my SpacePilot. If you can add that, then you can change it to _any_ combination you like :-)

Conclusion: I don't care much. Maybe #@$&!! in the beginning and then I get used to it...

lordofhyphens, 2018-06-14:
https://askubuntu.com/questions/118151/how-do-i-disable-window-move-with-alt-left-mouse-button-in-gnome-shell It's apparently a gnome thing.

We could probably move it to CTRL without any issue.

lordofhyphens, 2018-06-14:
Related for Windows: https://msdn.microsoft.com/en-us/library/ms971323.aspx#atg_keyboardshortcuts_keyboard_ui_design_rules

lordofhyphens, 2018-07-09:
@TheThirdOne Go ahead.

closed 2018-07-09 by lordofhyphens