slic3r/Slic3r #4447: Orbit cooling feature - but wait...

Huwyler, 2018-06-13:
Hi!

Please forgive me for coming up with another "orbit cooling feature" request. Although I never used orbit, I totally agree with you that it's not a good thing to do the way it has been done.

But I would like to ask for your opinion about a slightly different approach to such a feature.
Recently I installed a berd air cooling ring. As I understand, this differs from "traditional" cooling fans in two ways: First, the "fan" is strong, loud, heavy, but remote instead of moving with the print head. And second, the air is not blown underneath the nozzle but as an "air shower" around it. So, the cooling effect only takes place after the extruded plastic has firmly fused to the base. Normally, the "air shower" will pass by a fraction of a second after the plastic has been laid down. 
But not so if I have one single very small feature being printed (a pole with a diameter of 3mm, for example). Here, the air shower keeps missing the plastic, and the pole is happily heated up layer by layer. 
Now, what if we could tell Slic3r to include an orbit with the radius matching the blower ring's radius after each layer in these cases? Oozing wouldn't be a real problem as it would only take maybe half a second. Actually orbiting wouldn't even be needed, but just moving aside by maybe 5mm and staying there for half a second before going on printing, but orbiting would maybe result in a more "uniform" cooling.

What do you think?

Tinchus2009, 2018-06-17:
well, this kind of cooling has been discused a lot in other places. Most people think "orbiting" is actually not really usefull. One example is exactly the one you described. But also another example would be overhangs. Other things about orbiting is that actually the coling privded by "orbiting" depends a lot of some variables like: how much air you blow, the geometry below the air duct ans some more etcs. From my persoal exeprience, I have tested about 15 ring ducts... I tested them in real life useing smoke and I even used solidworks flow simulation in them. From ALL of them, I found 1, ONLY one wich actually was able to keep an even air flow 360 degrees around the printed area,  only 1... So in my opinion working in sucha  feature would be like a really waste of time for developers and would introduce a bunch of problems too, specially oozing. What about min layer time?

gege2b, 2018-06-17:
Hi @Tinchus2009 just for the record and sorry to be a bit out of topic, what is this design ? :D

Huwyler, 2018-06-17:
Hi @Tinchus2009 
These ring ducts you've tested, are they all with fans? Here, I absolutely agree with you. But the berd air is quite different, because it doesn't use a fan but an air pump. The pressure is MUCH higher. So, the airflow is quite uniformly divided to the air nozzles around the ring. I still have virtually no experience with it, but I really like the idea behind it: If you blow air underneath the nozzle, you cool down the plastic before it's supposed to be fused to the layer below. It's really not about uniformity, but just about not to blow air below the nozzle. But as great the idea is, what I mentioned is a clear draw-back.
Again: We're talking about maybe one orbit in a half of a second or less, maybe just during retraction.

Tinchus2009, 2018-06-17:
I will answer yboth @gege2b and @Huwyler as short as posible because I think this is not the place for a debate since this is meant to be used for slic3r development.
The design @gege2b is: https://www.thingiverse.com/thing:1025471 and also the second version of it. This cooler so far is the best one I have tested. It distributes the air perfectly even all around the 360 degrees. In fact I think is so good because actually creates an "orbit effect" around the noozle. Only disadvantage: you cant see the noozle. Believe me... I have literally printed like 20 different ring ducts, this is the only one that has a real engineering design.
@Huwyler entering into a debate here .. is not the place I think. I have tested radia, axial and pump version. About pumps:  the design of the fanless version made by sgabo is the one wich actually gave me the idea Im useing in my printer wich is a custom version of it. But you are thinking a litle too much simple and forgetting a couple of important facts: 1- air flow dinamics is not water flow dinamics and I have seen LOTs of ring designs claiming to be good because they tested with water... 2.- air flow is not a laser beam. That means that is imposible to direct the flow enterily to a specific point, without affecting nearby areas. 3- the air flow actually becames somehow complicated when you see in simulation that the flow direction gets affected by the same flow comeing from other direction because of the ring design and this is when the design comes into the game: the design has to be really good to make the air convergence be something accurate  if not, air flow will move in such a way that actually you get cooling effect only in 1 side of the 360 degrees area. That is why you see in a circular overhang cooling test that some areas of the overhang are printed ok, and some not, it is because some areas are better cooled than others, and then people ask themselves :why? if my duct is circular...?"
4.- You are thinking that a pump is used because of the pressure of air but this is actually not completly true. Pumps are used because you need pressure to overcome all the hose lenght you have. So you need pressure to be able to take enough air from the pump to the ring duct. And pumps are used because... they are cheaper than blowers! In my particular design, I had a blower fun, a big one wich I use instead of a pump (I have a big acuarium pump to). That blower is powerfull enough to let me use it at a max 50% of its power, it blows a LOT of air with more than enough pressure to overcome 1 mts of flex hose. Adn the advantage is: is silent. If I would have to use the air pump it would be a nightmare.
And my final thought about orbit cooling: all people talking about orbit cooling is like are forgetting something: air flow dinamics. I will explain shortly: imagine an air stream comeing out of a hole of a ring duct or any other kind of duct. 40% (sometimes less) of all the air comeing out of it will be directed in a stright line and like 60% (sometimes more) will be difused to the sides. How much to the sides? like something around 45 degrees. If that hole is 2mm above the printbed that means that you will have 40% of the air hitting your central point and 60% of the air cooling an area3 to 4 mm away of it... and what is this? ORBIT COOLING! So actually of the design of the duct is ok and the mounting of it it is ok (THIS IS SUPER IMPORTANT) you already will have "orbit cooling". The design https://www.thingiverse.com/thing:1025471 actually achieves this: it creates a bubble of air right below the noozle. A bubble means that air forces in the center cancel each other ccreating a bubble of no air flow and cooling happens for a low pressure area created there. Adn the air then goes all around.  Again , this ring design is the best I have seen. The concept created by SGABO is very interesting because it lets you remove the blower/fan from the printhead, so you have less weight there and also the flexible arms lets you aim the air very precisely. 

From my point of view, cooling is a matter of hardware  and should not be "fixed" by software.

Huwyler, 2018-06-18:
Hi Tinchus,
in one point I don't fully agree. That this is not the place for a debate about cooling. In this thread I asked about other's opinion about orbit cooling in the context of some type of cooling principle that MAY become very widely used. And you give (good!) arguments against this. I think, this is the perfect place for it.
I won't go into everything you wrote. I can very well see your point and I understand that you have done tremendous investigations on that topic.
I surely know that you need much more pressure to overcome the pressure drop in the tubing. What I meant was that the ring itself is still much more pressurized than with typical fan-based ducts, which minimizes the influence of the pressure drop within the ring. Some ducts (like the one you mentioned) decrease the cross circuit of the duct to take care on that, but many just ignore this, resulting in a design where half of the ring renders just useless as no air will ever blow from the far sided holes. So, by "uniform distribution" I just meant that the holes themselves behave much more evenly here.
Of course I know that the flow won't be laminar and not comparable to water. As the angle between the holes and the nozzle is quite flat (maybe 1 mm vertically and maybe about 5 mm horizontally), I thought you could still sort of think of it as an "air shower". But I can see your point about the influence of all these air flow components to each other. 
Another thing you mentioned was overhangs. Yes, I think I'm getting the point here. With overhangs some cooling is actually needed at the very moment the plastic is fused to the layer below.
Could yo tell us something that custom design of yours?

Tinchus2009, 2018-06-19:
Yes... when I said this was not the place I wanted to not upset the moderators lol, sice this place should be for issues of slic3r je. But yes, exchange of ideas is somethign good.
About cooling overhangs... well, this is only a though but for me, all the cooling investigation started with overhangs because I was haveing problems when I stared 3d printing with curling at overhangs... those cursling made me loose lot of prints because the noozle eventually  hit them and the motor looses steps.... so one day I stopped printing and focused about that specific solution. The best cooling I think is the one wich focuses air stream around the tip of the noozle, not exactly in the tip. If you achieve that you will cool the plastic inmedaitle after is layed down, and the plastic comeing out of the noozle will be cooled by an indirect stream. Right now Im useing 2 systems and of both of them Im like useing more the second one (the fanfan duct, sligktly modified). system 1 is the fanless cooling system from sgabo: https://www.thingiverse.com/thing:1627692
I dont know why that aprouch was not used before but it is great. Main advantage is the ability to precisely point the air stream to the exact place where you need it. The modification I did to that system was replaceing the air pump for a big blower fan. That gave me the ability to control de amount of air through the software since you can not do tat with an air pump (not unless you know about electronics and make some pcb to adap it). I used that system for about 1 year. From time to time I checked new ducts designs. Till I saw the fan fan (https://www.thingiverse.com/thing:2175956). I modified the angles a little (the original design blows too much into the noozle tip to my taste). But this is a simple a good design since lets you direct the air right next to the noozle tip and creates a bubble right under the tip). The overhang test I showed you was done with this alst system and it is the one Im useing most  because it lets me use a small blower fan and it is really silent. With goo tunning, this system is so precise that I can use cooling while printing ABS with NO WARPING at all since I can use really small amounts of air (in my case 7% of the blower power). You would go crazy if you see the bridges I can make in abs and its quality lol. I have achieve 15 cms long bridges with only 0.5 mm sagging in the center! this is only posible thanks to the precise amount of air aimed to the correct point in very small amounts.
A problem I see in design too is that many people think they work because they just increase the fan power and so the amount of air till they get good result and actually they are masking a bad air aiming. Te air is not going to the correct point and they mask this probem just sending more air. Then the cooling works because of the air getting to the correct point just because SOME of all the air is getting there, the rest is wasted. So Far the fanfang system has let me use a max of 70% of the blower power (and that was for the extreme overhang test, but in real life I have never used more than 50% wich would be in bridges)
Hope all this info helps you.

One interesting setting I requested here some time ago was the ability to control the cooling fan through  a modifier or haveing the option to set specific % of cooling at overhangs areas. Such settings would give us the abilitiy to do that: have control over the cooling of overhangs.