slic3r/Slic3r #4452: Raft first layer not being displayed

Tinchus2009, 2018-06-20:
### Version
lastest dev version

### Operating system type + version
win10 64b

### Behavior
I tested with different object so I think I can say in not an object dependant issue.
When you configure a raft and go to the 3d sim view, the first raft layer is not there (it prints, but it is just not shown in the 3d view).
In this image you can see the forst layer, it shows empty, but actually there it should be the forst layer of the raft. It is there in the gcode.

[image]

The second layer it shows ok:

[image]

foreachthing, 2018-06-20:
Nope. There it is ... starting at the first layer:

[image]
(The green stuff is the raft. Set to 5 raft-layers.)

What about the Layers-tab? What does it show there, on the first layer?

---

**Wait!** If set to ONE raft layer, it wont show it.
If set to two, only one layer will be show (on the **first** layer!).
If set to >=3 raft-layers, all of them are shown.

Tinchus2009, 2018-06-20:
mmmm we are getting different results now. I tried (to be sincere I have never used a 1 layer raft je). Setting 1 lyer raft: it shows ok. setting 2 and 3, still showns ok, setting a 4 layer raft then the trouble start and the first layer raft is not shown

foreachthing, 2018-06-20:
> we are getting different results now

AWESOME! :-D That makes it so much easier to find the issue - NOT!

Seriously now: what's going on here? 
Well, in Slic3rs' defense, I was using Slic3r-master.2018.06.09.1716.c058ae9.64bit.

[after a while] 
... ok, so it doesn't make a difference with Slic3r-master.2018.06.18.1719.354b623.64bit. Still the same behavior as described above.