slic3r/Slic3r #4453: Brim created on top of raft/in the air

Tinchus2009, 2018-06-20:
### Version
latest dev version

### Operating system type + version
win10
### Behavior

I tested this with some objects and the issue is on all of them so I can say is not an object dependant issue.
Problem: the raft has a size iqual to the object. Sometimes you need to increase the size. To the moment is not posible so a solution would be to use a brim to give more support for the raft? So if you activate the brim, it is created around the object but only after the last raft layer, so it ends in the air.
It is not very clear but  you can see there the brim in the air

[image]

on the bottom right you can see the layer height, the raft there was configured to be of 5 layers, brim is created at layer 5

[image]

lordofhyphens, 2018-06-20:
I believe this is a dupe.

lordofhyphens, 2018-07-11:
It is also, so far as I can tell, a visual bug and not in the exported GCode.

Duplicate of #4034. Closing.

closed 2018-07-11 by lordofhyphens