slic3r/Slic3r #4456: Unable to upload in an octoprint folder

4rnoP, 2018-06-20:
### Version
Slic3r 1.3.0

### Operating system type + version
Fedora 28

### Behavior
* _Describe the problem_
I can't upload an gcode in a specific octoprint folder. If I write the path in the "file name" field, an error occurs.

* _Steps needed to reproduce the problem_
Try to upload any gcode using the "Send to printer" button (with a properly octoprint configuration), and write for example "caisson/coin-int.gcode" in the  "file name" field.
* _Expected Results_
The file is uploaded in the folder named "caisson"
* _Actual Results_
[image]

Tinchus2009, 2018-06-22:
But you cant do this. The upload to octoprint ability, just lets you write the file name, no its path. Octoprint handles this by himself. I you want to change the folder where ht gcode is uploaded, you have to do that in octoprint

tedder, 2018-06-24:
Not really true- [the API](http://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder) supports local and sdcard destinations, and a path (folder) can be specified.

lordofhyphens, 2018-06-25:
We probably don't support that from the API (yet). Feel free to submit a PR
to address that.

Tinchus2009, 2018-06-26:
@tedder "not really true"? that sounds a little rude dude... If you want a TRUE then let me clear some things for you:
1.- the API you are talking about is an octoprint API, so maybe you are in the wrong place?
2.- Read the API documentation carefully: the upload capability is for the HTTP interface/protocol, not the samba. If you are going to communicate to octoprint and you dont do it through http then the samba is used... and samba doesnt let you specify a path unless that path is already configured in the octoprint server
So... actually you statement is NOT TRUE.

lordofhyphens, 2018-06-28:
@4rnoP Slic3r currently does not support upload locations to Octoprint. As a workaround you could create the directory `/tmp/caisson` on your local computer (where Slic3r is running) and it would probably work as you were trying to use it. You'd have to remake the directory every time you restarted your computer, though.

@Tinchus2009 The Octoprint API clearly states that one can provide a path to a parent folder to upload files to. Slic3r currently doesn't fill that out. @tedder is correct in their statement. Additionally, the request/issue reported by @4rnoP is specifically referring to the Octoprint upload API; I don't know what you were talking about in your message.

For anyone who is interested in implementing this, the area of interest in the code is around https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm#L2305

tedder, 2018-07-04:
FWIW, it can probably be backported from Prusa Edition.

[image]