slic3r/Slic3r #4458: "default" "First layer" width is not 200% as stated in the tooltip

l29ah, 2018-06-20:
### Version
354b6230bdea90b50b1c422a8da94c1f59313954
The screenshots were taken on the earlier version, but the described behaviour is the same.

### Operating system type + version
Gentoo Linux

### Behavior
default:
[image]
200%
[image]

#### STL/Config (.ZIP) where problem occurs
[bug.zip](https://github.com/slic3r/Slic3r/files/2121380/bug.zip)

xoan, 2018-06-21:
When the width is a % it is computed over the layer height (as stated in the tooltip).

l29ah, 2018-06-21:
It doesn't contradict my reasoning in any ways, as the tooltip states:
`
Set this to a non-zero value to set a manual extrusion width for first layer. You can use this to force fatter extrudates for better adhesion. If expressed as percentage (for example 120%) it will be computed over first layer height. (default: 200%)
`
And i don't change anything else, especially the height that is the same according to the screenshots.

xoan, 2018-06-21:
OK, I misunderstood your issue.

"default" (as value) inherits the "Default extrusion width" specified in the first option.

The "default" in tooltip stands for the default value of this option in -default- profile.

l29ah, 2018-06-21:
Then it's strange since "default" turns out thinner than "200%" as seen on the screenshot.

xoan, 2018-06-21:
200 % 0.2 mm = 0.1 mm (this is the extrusion width of the first layer when 200%)

```; external perimeters extrusion width = 0.44mm (2.38mm^3/s)
; perimeters extrusion width = 0.56mm (6.20mm^3/s)
; infill extrusion width = 0.50mm (7.31mm^3/s)
; solid infill extrusion width = 0.56mm (2.07mm^3/s)
; top infill extrusion width = 0.56mm (1.55mm^3/s)```

And this are the values for the diferent paths roles the first layer inherits when "default".

lordofhyphens, 2018-06-28:
I can reproduce this. I think "default" in this case ended up being `0` which is shorthand for auto.

lordofhyphens, 2018-06-28:
12aa06e09c6460165f370ebd8e1b14295ca14290 should fix this. @l29ah please let me know (you may need to re-select default from something else to get it to stick).

lordofhyphens, 2018-06-28:
I also added `auto` as an option in the dropdown (which is what it had gotten changed to). 

Overall, looks like a regression to me when @alexrj refactored and updated the UI for the preset editors.

closed 2018-06-28 by lordofhyphens

lordofhyphens, 2018-06-28:
Yes, it's fixed now. You will have to change your config to set it to default instead of auto. 

[image]

lordofhyphens, 2018-06-28:
@l29ah A word of caution: 200% puts the extrusion width right at your nozzle diameter. The general use case for first-layer extrusion width defaulting to 200% is that it is expected that first-layer height is a value close to the nozzle diameter.