slic3r/Slic3r #4459: Build Plate Shown in Negative Area

LandSquidConfirmed, 2018-06-21:
### Version
1.3.1-dev
Windows 10

### Behavior
The plater shows the build plate in the negative areas of the coordinate plain, and slices the object as if it is there (Gcode has all negative values). This problem is occurring for a very large printer profile (2235x2159mm), and does not occur if I  decrease the size to 2000x2000. To reproduce the problem create a printer with a bed size greater than about 2200x2200. I was wondering if some small change in the ini files would work as a quick fix for this, or if it could be fixed in a future release. 

[image]

My configuration file: 
[ArmadaSlic3rConfig.zip](https://github.com/slic3r/Slic3r/files/2125283/ArmadaSlic3rConfig.zip)

The STL file used:
[Test Square.zip](https://github.com/slic3r/Slic3r/files/2125290/Test.Square.zip)

lordofhyphens, 2018-06-21:
The scaled value is probably overflowing.

LandSquidConfirmed, 2018-06-21:
I'm running the 64 bit version, so I'm not really sure why that would be happening unless the scale variable is an integer that gets converted to a float or something, because to my knowledge floats and doubles can't overflow in c++ and integers can get pretty big. Maybe if the integer were made unsigned the problem would be solved? Not really sure if that's feasible but...

lordofhyphens, 2018-06-21:
Unsigned isn't feasible because it's just as possible for someone to put 0,0 in the center of their bed.

LandSquidConfirmed, 2018-06-22:
Not sure if this is already the case, but is there any reason not to put?
if (64bit) 
int64 scale = ...
else
int scale = ...

lordofhyphens, 2018-06-28:
Well, I can't reproduce on Linux, so there's that.

[image]

lordofhyphens, 2018-06-28:
@LandSquidConfirmed that trick will work up to around 4000x4000, after which the 2D draw code apparently crashes because of a "division by zero".

lordofhyphens, 2018-06-28:
@LandSquidConfirmed I am suspecting the underlying draw code is what is being limited here, but I haven't looked at it closely enough to be sure.

Unscaling 2300mm yields a value of 2,300,000,000. Just a FYI.

supermerill, 2018-07-10:
internally, slic3r use int32_t, so the precision is 2^31 = 2 147 483 648
the "scale" is 0. 000 001 => if unit = mm; the smallest unit is on nm scale (but i won't count for anything below 100nm as rounding errors can propagate a bit).
So it should be able to use 2 147 mm * 2 147 max

In windows 32bit, i have a 'int' in stdint.h in visual studio.
gcc on linux 64b may set 'int32_t' to a 64 bit word?

supermerill, 2018-07-10:
@lordofhyphens  in libslic3r.h, line 41: there are an error: the type "long" is angobious, it can be a 32b word. You should replace it by int64_t.

lordofhyphens, 2018-07-10:
@supermerill good idea.

lordofhyphens, 2018-07-10:
Implemented as #4480 to test it out first (using intmax). Thus, you'll still run into this problem on a 32bit build of Slic3r, but if you're stuck using 32bit systems with Windows in 2018 I feel bad for you.

lordofhyphens, 2018-07-30:
@LandSquidConfirmed Does this sort the problem out for you https://ci.appveyor.com/api/buildjobs/oktgsia2k7bocco3/artifacts/Slic3r-fix_int.2018.07.30.1796.8f32688.64bit.zip