slic3r/Slic3r #4460: Building Slic3r on Ubuntu fails: Slic3r::FillGyroid::FillGyroid()«: Fill.cpp FAIL

Annaf12, 2018-06-22:
### Version
VERSION = '1.302136'

### Operating system type + version
Ubuntu 18.04 LTS

Hello

I want to work with Slic3r and tried setting up Slic3r several times. The build was successful and after that I wanted to create an executable file by using CMakelists.txt but it always fails with this message
[image]

I have not changed anything in the source code. I got it from Github and wanted to create an executable file without any editing. Can someone give me a hint, why it does not work?

Thanks in advance and best regards

Annaf12

Samir55, 2018-06-22:
Isn't this fixed by #4451? Slic3r builds successfully on my machine. Please check your repository is up to date.

lordofhyphens, 2018-06-22:
First, read the wiki about building on Linux.

What you are building is a small program that is made for slicing to SVG (for DLP printing).

It is not the main application.

lordofhyphens, 2018-06-22:
https://github.com/Annaf12/Slic3r/blob/master/src/CMakeLists.txt certainly looks like it needs to be updated if indeed you wanted to build that application.

lordofhyphens, 2018-06-28:
Can't reproduce. Closing.  Also, that version number is suspect--there is no version 1.302136 of Slic3r.

@Annaf12 If you can provide the git commit you are working off of via `git rev-parse --short HEAD` and it's still failing for you as of  354b623 I will reopen.

closed 2018-06-28 by lordofhyphens