slic3r/Slic3r #4462: Settings tab open by default

exuvo, 2018-06-24:
### Version
1.3.1-dev

I would like the settings tabs to be open by default when the program starts. I really don't see the point with having to open them all the time.
A simple checkbox in File>Preferences would suffice.

mdealer, 2018-09-28:
You don't need any settings for 3 tabs...

szernex, 2018-10-31:
I also second this, in earlier versions I remember this being the case, but at least in the latest 1.3 dev releases it no longer is. Don't really see the reason for removing them, only makes the user have to click 3 cogs every time they open the program and they need to adjust settings.

alranel, 2018-11-06:
The original reason was to encourage a usage model where users would not touch Print Settings on a per-print basis, but they would only tweak the settings exposed in the right-side plater panel. The Print Settings tab/dialog would be needed with much less frequency.

For some reason, it seems that all of us (including me!) are still touching Print Settings daily.

I do actually agree with your proposal of making them fixed again (no need for an additional checkbox: we already have one that opens them in dialogs instead of tabs)... but can you help me understand why the aforementioned model doesn't work for you? Did you actually try exposing the most frequently used options in the plater, and reducing the need for opening Print Settings?

Bluelinegecko, 2018-11-06:
I can see your reasoning for the change, but I feel like there should at the least be an option in the "preferences" menu to have them always open.

I have numerous different profiles for printing and often might need to tweak just one or two settings for a specific print. I don't want to go through that many profiles and set which "shortcuts" I want visible. I just want to be able to see all the settings I can at one time and change what I need.

There is limited space in the "shortcut' window on the platter screen and I'd much rather just have access to everything in  the print setting tabs. It's not any extra work if they are already open by default.

alranel, 2018-11-06:
Thank you for the answer, that's what I wanted to hear. Basicaly, you're saying that settings shortcuts might work for you but it's too much work. And this makes sense.

I think we can make the tabs fixed again (unless the option to show them as dialogs is selected).

closed 2018-11-10 by lordofhyphens

lordofhyphens, 2018-11-10:
Done. They are now fixed tabs unless you made them dialog boxes via Settings.

commit 59164f7b (2018-11-10): If tabs are used, make them fixed instead of appearing/disappearing.
Fixes #4462