slic3r/Slic3r #4464: "Move t bed center" somehow not properly working

Tinchus2009, 2018-06-28:
### Version
latest dev version

### Operating system type + version
win10 64b

### Behavior

Load any model, then select it, and chose from menu "Move to bed center", the object will be placed at the bed center, then slice (selecting the .preview tab). You will see that the sliced preview will not be in the bed center but in the original object location. The 2D view showd it in the bed center, the preview doesnt, and the layers view shows the layers in the object's original position (not in the bed center)

[image]

Moved to bed center:

[image]

Then the sliced preview:

[image]

lordofhyphens, 2018-06-28:
@Tinchus2009 "latest dev version" doesn't mean much, please include at least the short commit hash.

If you're using files off of dl.slic3r.org/dev/win, it's the number right before "32bit" or "64bit" in the filename. 

If you have a git checkout, `git rev-parse --short HEAD` will give the string.

Tinchus2009, 2018-06-28:
Sorry, it is this one:

354b623
-- | -- | -- | --

lordofhyphens, 2018-07-02:
Fixed via #4465

closed 2018-07-02 by lordofhyphens