slic3r/Slic3r #4466: variables and placeholders

GuismoW, 2018-07-02:
### Version
_Version of Slic3r used goes here_ : Slic3r Prusa Edition 1.14.0-beta+-win64

Hello, I'm pretty new in the 3D Printing World (and first time I use GitHub - if need a presentation, please let me know)
I'm using Repetier with Slic3r Prusa Edition.
First off : nobody tell the difference between the Prusa Edition and the Slic3r (simple) edition,
it would be great to have a sum-up. 
I recently slice a Temp Tower, and the Prusa Edition failed : was unable to use the same speed for the same time (external perimeter) whereas the simple edition worked correctly, as well as Cura.

### 
I have request (hope it's the good place to write down) : beeing able to have variables, or use math with placeholders

When heating up the nozzle, filament comes out (normal). 
I don't want to activate Skirt/Bim, 
but  I'd like to make my own wipe motion before the real print starts

here my code in "Start G-Code" (repetier / Slic3r Prusa Edition 1.14.0-beta) :
G28 ; home all axes
G1 X150 Y150 Z50 F10000
;G1 E110 F9000
M109 S[first_layer_temperature_[current_extruder]]; set temperature and wait for it to be reached
G90
M82
;wipe the extra filament from the nozzle, because heating up the nozzle extrudes filament
G1 X10 Y10 Z[z_offset+first_layer_height] F2000
G1 F600
G1 X60 Y10 E10
G1 X10 Y[10+extrusion_width] E5; 
G1 Z2 F2400

* Results
23:40:28.072 : <Slic3rPE> G-code export to composition.gcode failed due to invalid custom G-code sections:
23:40:28.072 : <Slic3rPE> 	start_gcode

I have a glass on the bed, and the z-offset is set to 1mm
but if I want to slice with another printer where the z-offset is different, I need to change the code manually, 
it would be great if we could use placeholders like this : 
      [z_offset+first_layer_height]
      [10+extrusion_width]

What do you think of such feature ?
(if I need to move the thread, please tell me, I'm new on the github)

lordofhyphens, 2018-07-02:
Sorry, we can't support the Prusa3d fork here.

closed 2018-07-02 by lordofhyphens

gege2b, 2018-07-02:
hi there
Differences between Slic3r and Slic3r Prusa Edition is simply that the prusa edition have features and tweaks targeted to their printers, they can also implement whatever they want and how they want, Slic3r "original" is more printer-agnostic

By the way, this is the Slic3r original (or legacy, whatever) github, There is a [github for Slic3r PE](https://github.com/prusa3d/Slic3r) witch you should use for issue concerning their fork

lordofhyphens, 2018-07-02:
However, what you want already exists in main Slic3r: https://github.com/slic3r/Slic3r/wiki/Conditional-Gcode-Syntax-Spec

While I think the Prusa3D team has implemented their own version, I believe that the syntax may be different.

lordofhyphens, 2018-07-02:
@gege2b we're Slic3r. They're Slic3rPE. 

Legacy implies nobody's working on it at all :p

gege2b, 2018-07-02:
yeah, pretty bad choice of word sorry :)