slic3r/Slic3r #4467: Only plater tab shown after building on arm exynos cpu

theDoktorJ, 2018-07-02:
### Version
1.3.0-18-g65bcaff6
Git Commit ID:  65bcaff615b3ff57d279cabd58ec588a22d277ff

### Operating system type + version
Ubuntu Mate Release 18.04 LTS (Bionic Beaver) 32 bit
Kernel Linux 4.14.52-145 armv7l
Mate 1.20.1

### Behavior
I successfully built slic3r using this tutorial:  https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux.  Then I created the slic3r launch script as described in the tutorial and made it executable.  Then I launch it using ./slic3r from the directory where I built it.  It launches, but only the Plater tab is visible, and I get the following errors:  
odroid@odroid:~/3dPrinting/software/Slic3r/src/Slic3r$ ./slic3r
MESA-LOADER: failed to retrieve device information
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos

[image]
[image]

#### STL/Config (.ZIP) where problem occurs
I don't even get to the point of adding an STL before the problem occurs.  The attached config is simply what got automatically imported from slic3r 1.2.9 which works properly.  No changes were made to the config in the git version of slic3r.  In fact, I can't actually make changes to the config in this version of slic3r since the tabs are missing.  
[config-2018-7-2.zip](https://github.com/slic3r/Slic3r/files/2157012/config-2018-7-2.zip)

lordofhyphens, 2018-07-02:
Your opengl apparently is misconfigured or something, because that's what the error means. 

Check to see if a build from dl.slic3r.org/dev/linux works

theDoktorJ, 2018-07-03:
Yeah, that seems to be the case.  I was thinking it was a bug caused by my unusual arm platform, but now that I know it's just a UI change, I'm fine with it just like it is.