slic3r/Slic3r #4468: Setting up a custom infill pattern

jclinton830, 2018-07-03:
### Version
1.29

### Operating system type + version
Windows 10

### Question
I have been trying to set up a custom infill pattern for a bio-printing application within Slic3r. The problem I am having is that after I have written the c++ class that does the infill within libsli3r/Fill/, even though I am able to select this new infill pattern in the infill settings section in the GUI it always executes a rectilinear infill pattern. 

Is there something else I need to add/edit for slic3r to graphically represent my custom infill in the GUI?

lordofhyphens, 2018-07-03:
No, but rectilinear is the default for top, bottom, and all internal solid infill.

lordofhyphens, 2018-07-03:
Post a link to your work if you can with screenshots and I (or someone else) can take a look.

jclinton830, 2018-07-03:
Hi @lordofhyphens!

Here is the link to my repo I set up to share our work with you.

https://github.com/jclinton830/customSlic3r

The files you might be interested in would be called FillBioRectilinear.cpp and hpp within xs/src/libslic3r/Fill folder.

The intention was to create an exactly same rectilinear pattern but with controllable line spacing functionality between the each rectilinear line. A parameter called fill_spacing is what the user changes to control this line spacing. This variable is declared in Fill.hpp.

Appreciate your help. 

Thanks,

Jerome.

lordofhyphens, 2018-07-03:
https://github.com/jclinton830/customSlic3r/blob/8ebaeae26c751b8a76a64613fab1d1713930dc6e/xs/src/libslic3r/Fill/FillBioRectilinear.cpp#L20

Looks like you have no methods implemented for your new fill. I have no idea how it even compiles.

Thoughts for your use case: 
* What are the expected units for fill_spacing?
* How is fill_spacing intended to interact with infill_density and the infill extrusion width (wider lines normally create larger patterns for the same infill_density)?
* fill_spacing probably should be defined in PrintConfig and made available in PresetEditor for users to modify -- it would be a lot faster to tweak than recompiling a constant that way.

Development flow questions:
* Were you planning on submitting a pull request to add this to main Slic3r? Looking at your source, this is going to require that you properly fork Slic3r on Github, make a branch, copy your changes over into that branch, and then make a pull request.

lordofhyphens, 2018-07-03:
Wiki entry on adding new items to the GUI/CLI: https://github.com/slic3r/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI

lordofhyphens, 2018-07-03:
Also, you should probably work from 1.3.1-dev (current HEAD) or start your branch from the 1.3.0 tag if you want to start "from a release". If you *really* want to baseline off of 1.2.9 (I wouldn't), you should do a git checkout of the 1.2.9 tag, make that a branch, and do your work there. 

Reference:
https://stackoverflow.com/questions/10940981/git-how-to-create-a-new-branch-from-a-tag

Oekn5w, 2018-07-03:
@jclinton830 take a look at #4317, where a new infill pattern was merged, about where you might have missed something

jclinton830, 2018-07-03:
@lordofhyphens, thank yo for taking the time to look at this. 

Sorry it is a bit difficult for me to understand what is going on in Slic3r source code, because I have never used perl before. Anyway, l will answer all your questions below. 

Thoughts for the use case:

* The fill_spacing is meant to be in millimeters.
* The idea is to have the extrusion path density to vary as it normally does in regular slicing, but just control the distance between each path (center to center distance of each path). I do not want this controllable spacing distance to affect how the infill_density or the extrusion_width behaves.
* Please look [here ](https://github.com/jclinton830/customSlic3r/blob/8ebaeae26c751b8a76a64613fab1d1713930dc6e/xs/src/libslic3r/PrintConfig.hpp#L252)for fill_spacing variable's declaration in PrintConfig.hpp and [here ](https://github.com/jclinton830/customSlic3r/blob/8ebaeae26c751b8a76a64613fab1d1713930dc6e/xs/src/libslic3r/PrintConfig.cpp#L507)for the min and max values setup for the user to select in PrintConfig.cpp
* fill_spacing has also been setup so that the user can select a value between a range within [PresetEditior.pm](https://github.com/jclinton830/customSlic3r/blob/8ebaeae26c751b8a76a64613fab1d1713930dc6e/lib/Slic3r/GUI/PresetEditor.pm#L555)

Development flow questions:

*No, this is just going to be used for one experiment that is part of a research. Therefore I do not think it is necessary to add this as a feature to slic3r, also do not believe it would be of any use to the majority of the slic3r developers and users.

If I have not answered all your questions well please let me know. 

Thanks.

jclinton830, 2018-07-03:
@Oekn5w Thanks for that. I will have a look!

jclinton830, 2018-07-04:
@lordofhyphens After some testing, turns out that my code is actually working like expected. I think I just wasn't adjusting a parameter like I should have for it work. 

I am successfully able to control the infill path spacing between extrusions now. 

Sorry if I have wasted your time. Thank you and appreciate your help though.

closed 2018-07-04 by jclinton830

lordofhyphens, 2018-07-04:
Glad to hear it is working for you.

shamshx, 2020-10-07:
@jclinton830 can you please share your source files for this? I'm interested in a similar application for my school work and really need your help.