slic3r/Slic3r #4469: Thin wall : improvements

supermerill, 2018-07-04:
Hi,
I just slain my personal dragon by improving the thin wall algorithm.

 * no more strange turn at end => fusion of voronoi end lines when they are near enough
 * anchoring inside the perimeters => extends the lines end inside the perimeter to join it. A gui parameter can be created to adjust it if needed (currently, it extends to meet the center).
 * conservation of too small segments of thin walls when they are crossing each other => modification of the fusion algorithm to select the most strait path instead of a random one
 * no more cutting of small walls path by the gcode writer (no_sort for each thin wall extrusion) to have nice long lines of extrusions instead of many smaller.

before:
[image]

after:
[image]

My problem: Last time i tried to merge my code in alexrj branch, it was a chore for me (bad at perl and enormous time for compiling)

But perimeterGenerator.cpp and ExPolygon.cpp seems close enough to create a pull request for a sub-part of my modifications (point 1->3). For the "no more cutting of small walls path by the gcode writer", i won't translate my c++ code into the perl codebase (that's why I didn't submit a pull request for my ironing code).

It is ok?

lordofhyphens, 2018-07-04:
@supermerill if you don't want to port your C++ ironing code to perl, submit a PR for just the C++ portion and I'll cook up an XS so it can be called from the perl section. Or port more of gcode to C++ :)

lordofhyphens, 2018-07-04:
As for compiling, there's a trick you can do so long as you're done changing stuff in libslic3r.h:

On Linux or OSX: 
`eval $(/path/to/perl -Mlocal::lib=/path/to/slic3r/local-lib)`

On Windows Powershell:
`Invoke-Expression "$(perl -Mlocal::lib=/path/to/slic3r/local-lib)"`

Either command will set up things in that CLI window so you can just cd to xs and run `perl Build.PL` and `./Build install`. That will just build the XS/C++ part.

I am de-Perling Slic3r as fast as I can manage.

supermerill, 2018-07-04:
Thanks, i will try.

lordofhyphens, 2018-07-04:
@supermerill I'll be on IRC quite a bit over the next few days (including today) if you want to chat in realtime about it. I'd love to get your code into Slic3r.

lordofhyphens, 2018-07-05:
Merged, thank you for the work @supermerill.

closed 2018-07-05 by lordofhyphens