slic3r/Slic3r #4476: Overhangs & perimeters in briged area

supermerill, 2018-07-09:
Hi
I was creating some options to let me throw badly made stl for 3D printing and having it not fail that much.

1) I added to "Extra perimeters if needed" the reverse: it add perimeters if possible instead to drawing bridge when there are overhangs : it check if the next perimeter is over a not-supported area and it ask to add one more perimeter in this case (repeat).

2) I added a feature that remove the detected "bridge" area from the perimeter generator (only the area that can be bridged, not the overhangs). With this, no perimeters are draw in these area (see pics for a use-case)

### **Extra perimeters if needed :** 
[image]

### ** Avoid unsupported perimeters :**
[image]

### ** An other exemple of Extra perimeters & a combination of both :**
[image]

[image]

What do you think of it?
What do you think it should do? (with an image please)
Do you have some torture test?

I will do some print of these "torture objects" to compare my changes with the current version.

my torture test object source:
[test_overhang.zip](https://github.com/slic3r/Slic3r/files/2177492/test_overhang.zip)

lordofhyphens, 2018-07-09:
@supermerill if you open up a PR, the appveyor build server will build your changes.

Here's the list of all of the appveyor builds:
https://ci.appveyor.com/project/lordofhyphens/slic3r/history

To get the artifact for just that build, you click into the individual build log and then click the "Artifacts" tab.

VanessaE, 2018-07-09:
You may be interested in #3950 also, @supermerill

supermerill, 2018-07-09:
@Tinchus2009 
You're right. My third test is just an impossible print without supports.
But for the first and the second, is it better?

Tinchus2009, 2018-07-09:
Yes! Actually I think you are thinking smehow in the same thing @supermill than a coupe of us. You are thinking about bridges but somehow the idea you are thinking about is actually overhangs
Right now a missin parameter  to better deal with overhang would be the abilitu to control the overhang speed and the fan speed at overhangs. Others features requested in #3950 would be awesome but I know that would be a lot of work to :)

supermerill, 2018-07-10:
Some tests:
wade with natural pla. Maybe a more opaque plastic should be better to see the differences.

[image]
[image]
[image]

Tinchus2009, 2018-07-11:
Please dont take my comment in a bad way, I really respect the effort you are doing. haveing said that: I dont really see any improvements in the print I see. From my point of view about what I see, is like you are trying to find a way to print areas that need supports. In your first pic you show an example of a perimeter belonginf to a hole. Well, that hole can be printed with no problem useing supports. I the pictures... I can not say much because they are not very clear to me, but what I can see is no really a big difference. I would also ask you like Vanessa ask you: to take a look at feature request #3950 because right now you are already in that issue. If you already made a code to separate bridge perimeter from perimeter geenrator, you could also make a code wich separates overhang perimeters, and then we could work on those "isolated overhang perimters" in order to control features like: their speed, the cooling on them, etc.