slic3r/Slic3r #4477: Crash when changing presets while current preset is modified but unsaved

dwillmore, 2018-07-09:
### Version
Current master branch as of 20180709 but has been in there for months.
Specifically: 1.3.0-23-gb3d3bed8

### Operating system type + version
Fedora F27 current as of 2018/07/09

### Behavior
Start with a fresh invocation of the Slic3r GUI.
1) Add a model--doesn't seem to care what it is
2) Change a setting under a Print Setting, Printer Settings, or Filament Settings
3) Without saving the preset (which will be marked as "(modified)"), try to change to a different preset of that type
4) crash "Can't locate object method "SetYesNoCancelLabels" via package "Wx::MessageDialog" at /home/willmore/src_new/slic3r/Slic3r/lib/Slic3r/GUI/Preset.pm line 91."

_Is this a new feature request?_
Nope, unless this is the desired behavior. :)

#### STL/Config (.ZIP) where problem occurs
I can add this if it doesn't reproduce with a default environment.

lordofhyphens, 2018-07-09:
Does work with AppImage build:

Expected behavior: MessageDialog popup 
[image]

dwillmore, 2018-07-09:
The appimage does provide that prompt.  Must be system dependent.  Removing system provided perl-wx and rebuilding with Slic3r provided/installed one.

lordofhyphens, 2018-07-10:
Closing as was related to individual build environment.

closed 2018-07-10 by lordofhyphens