slic3r/Slic3r #4478: issue  #2853 revisit for special application

Mettlerfreund, 2018-07-10:
### Version
Slic3r-console 1.2.9 (same with Prusas 1.40.0 etc) as part of a custom work flow for two-photon-polymerization (for a rough idea see: https://en.wikipedia.org/wiki/Two-photon_absorption)

### Operating system type + version
work flow on Windows 7-10
Slic3r-console is used to generate slices in the .svg-format.
Using clipperlib (by Angus Johnson) very much like slic3r does, I make special Infill and generate special G-code for printing with the two-photon-polymerization technique. I chose slicer for the slicing, because of its power and community support. I don´t want to reinvent the wheel and still have a robust slicer for many different printers in the lab.

### Behaviour
Issue https://github.com/slic3r/Slic3r/issues/2853 revisit: For my purpose I tend to print objects with very “large” dimensions (mm) but still very small substructures (nm). Therefore I interpret 3D-object-units as micrometer so that a cube 1000x1000x1000 is 1mm**3. Objects one magnitude larger than about 1000 Units are generally not accepted by the slic3r. Changing the printer-bed-settings does not change this behaviour. This limitation can be coped with by a scaling prior to the slicing as suggested in Issue #2853 like this:

>`slic3r-console --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 10mm.stl`
`=> Processing triangulated mesh`
`No layers were detected. You might want to repair your STL file(s) or check their size and retry.`

>`slic3r-console --scale 0.1 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 10mm.stl`
`=> Processing triangulated mesh`
`Exporting to 10mm.svg...Done.`

Afterwards the data is scaled up again and yields four slices.

I am very unhappy with this solution and don´t see a reason behind this limitation. Slic3r is using Clipperlib and Clipperlib is handling such dimensions just fine. 

The next order of magnitude (a box of 100mm**3) freezes Slicer entirely when I try the workaround with the scaling like:

>`slic3r-console --scale 0.01 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 100mm.stl`
`=> Processing triangulated mesh`

no error, just nothing happens

 What am I missing?
I really appreciate your help and hope to resolve this issue soon. My goal is to find a smart way to slice the object “100mm.stl”. It must be possible.

#### STL/Config (.ZIP) where problem occurs
the two stl.files and the configuration are in the zip:
[slic3r_issue.zip](https://github.com/slic3r/Slic3r/files/2180619/slic3r_issue.zip)

Cheers

lordofhyphens, 2018-07-10:
I'm sure that this also affects 1.3.0, but we do have 1.3.0 and you should probably not be using 1.2.9 :)

I wonder if this is related to #4459.

I suspect that you're running into problems with scaling -- any number used internally by Slic3r is going to be scaled (through division by 0.000001).

lordofhyphens, 2018-07-10:
When you can, could you test on Linux or OSX and see if the problem persists there for your workflow?

lordofhyphens, 2018-07-10:
@supermerill I've got a patch as #4480 that uses intmax_t instead. 

I think it is better to reach for the best on all 3 platforms and understand that there may be limits; we still want to support 32bit systems (within reason).

lordofhyphens, 2018-07-10:
nvm, intmax_t isn't that useful when I have to deal with prototypes that are looking for data types out of Clipper.

Mettlerfreund, 2018-07-11:
Thank you for the fast answers!

### I tested the same commands on a:
*OpenSuseLeap 42.1 System
*KDE Plasma Version 5.5.5
*Kernel Version 4.1.39-56
*OS Type 64-bit
*Processor 2xPentium Dual Core CPU T4500 @2.30GHz
*1.9GiB Ram
*slic3r-1.3.1-dev-b3d3bed-linux-x64.tar.bz2 from the slic3r Homepage

#### no scaling:
>`chmod 775 ./Slic3r`
`./Slic3r --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 10mm.stl`
`=> Processing triangulated mesh`
`Use of uninitialized value in subroutine entry at /home/max/Desktop/Slic3r/local-lib/lib/perl5/Slic3r/Print.pm line 173.`
`Exporting to 10mm.svg...Done.`
`Warning: the supplied parts might not fit in the configured bed shape. You might want to review the result before printing.`

Does generate around 15'000 slices with a z-distance of 1unit.
(Not part of this issue: the commands for the layer-hight and nozzle-diameter are overwritten for any value larger 1.) It does work!

#### scaling:

>`./Slic3r --scale 0.1 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 10mm.stl`
`=> Processing triangulated mesh`
`Use of uninitialized value in subroutine entry at /home/max/Desktop/Slic3r/local-lib/lib/perl5/Slic3r/Print.pm line 173.`
`Exporting to 10mm.svg...Done.`
`Warning: the supplied parts might not fit in the configured bed shape. You might want to review the result before printing.`

Does generate around 1'500 slices with a z-distance of 1unit. Works as expected!

#### stl-file one order of magnitude bigger:

>`./Slic3r --scale 0.01 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 100mm.stl`
`=> Processing triangulated mesh`

...and it never stops.

@supermerill : I too think that there should be no problem until very very large structures. Down-scaling the 3D-objects with the command `--scale` should solve the upper bound-problem for dimensions bigger than  2 147 units, but that does not seem to solve it.

Cheers
M

lordofhyphens, 2018-07-11:
@Mettlerfreund can you post a separate issue for the "never stops" with the reproduction steps + config? I'd like to investigate that separately. I suspect it's down in admesh or TriangleMesh (or the Perl conversion methods). 

Please create another issue for the CLI config override as well.

And yeah, I think the problem is how `long` is defined on Windows alone. Properly working around this is going to be a different headache.

Mettlerfreund, 2018-07-13:
### I conclude:

As for right now 3D-Objects with large dimensions have to be sliced on Linux to go beyond the limitation of 2147 units in x or y direction, or -on windows- they have to be down-scaled prior to the slicing operation (using slic3r on commandline with ´--scale´) and the output up-scaled with some script etc. For very large dimensions (Coordinates in the range of 10E5) Slic3r runs into problems (See #4486)

 The resolution on windows ist roughly limmited to 10E-5 %. On Linux it should be much higher (if you don't scale).

closed 2018-07-13 by Mettlerfreund