slic3r/Slic3r #4481: Cut operation should not move part to Z0

surdu, 2018-07-10:
### Version
1.3.1-dev

### Operating system type + version
Windows 7

### Behavior
This is a feature request.

I believe that it would be beneficial if the `Cut` operation window should offer an option not to move the part to Z0, and leave it where would remain after cutting it.

This will allow to print parts with different materials based on Z height, similar to Simplify3D's `Start printing at height` / `Stop printing at height`

Another use would be to resume a failed print by cutting an object bellow where the print failed and start printing from there on.

#### STL/Config (.ZIP) where problem occurs
Not applicable

lordofhyphens, 2018-07-10:
Why aren't you using modifier meshes to set extruders for changing materials?

lordofhyphens, 2018-07-10:
Fixing failed prints is enough of a one-off that I would suggest (ab)using the z offset option in Printer settings.

surdu, 2018-07-11:
Unless I'm missing something, I am not using modifier meshes because I don't have multiple extruders. The plan was to print the bottom part with one color, change the filament, and then print the second one on top of the first one.

lordofhyphens, 2018-07-11:
@surdu The usual workflow looks like this for what you are looking for (maximum flexibility).

1. Make a variant of your Printer profile with 2+ extruders. 
2. Set the toolchange gcode to be where you are prompted to change filament (and probably move the head out of the way).  
3. Act as if you are doing multiple extruders for much of the workflow.
4. Run a post-processing script that replaces/removes all of the `Tx` commands.

lordofhyphens, 2018-07-11:
Since you usually get bad fusing anyway when printing on cold plastic, you might as well just cut and print normally with the separate colors and glue them together (either with a plastiweld if they're the same material just different colors or superglue if they are dissimilar). 

You can usually find plastic cement like:
* https://plastruct.com/shop/glues-tools-and-accessories/00003-bond-2/
* https://plastruct.com/shop/glues-tools-and-accessories/2-ppc-2/

Both should work pretty well.

lordofhyphens, 2018-07-11:
I've added it to the feature request list if someone wants to submit a PR.

surdu, 2018-07-11:
I'll give that PR a try 😁

Tinchus2009, 2018-07-12:
Yuo dont need that to do a multicolor print: just print the object, make a pause at the desired point, change filament and continue... It is as easy as selecting "change filament " in your LCD interface, or inserting a m600 code at the point you want to change filaments... or you even have the multicolor print web interface at prusa website. There is no need to complicate things ;)