slic3r/Slic3r #4482: BRIM not coencted to object

Tinchus2009, 2018-07-11:
### Version

b3d3bed

### Operating system type + version
win10 64b

### Behavior
I have been observing something wrong with BRIMS: while printing ABS, the model just detachs from the brim, makeing the brim useless. It happened at least in 10 prints. I though something was wrong with my printer calibration but I checked everything and extrusion, movements, etc, all was ok. I stopped the print at layer 2 of a model and examind the brim: the conection to the BRIM is al st not there. I still though it was a problem on my side, so I fixed it somehow oeverextrudeing at first layer. But there is a problem there and I have just seen what it is: the brim is being sliced not conected to the object... if this is supose to be like this, I think you need to review this behaviour. Brim is supose to be well atached to the object i not, ABS prints wont work. Brim was working ok at least 3 or 4 releases ago, really cant tell how I started to see this behaviour

In the screenshot you can see: brim is on the right, it is not conected actually to the object

[image]

lordofhyphens, 2018-07-11:
#2230
https://github.com/slic3r/Slic3r/commit/af46db320f820cc68e692cccf2f83890f4ee1596

Tinchus2009, 2018-07-11:
Oh.... I can see that I already made a comment in that issue... I didnt knew this was already implemented. Well... My opinion is based on real life observations now. Printing ABS now useing slic3r is almost not posible. There is no way you can print ABS without haveing the help of a brim. And right now, the brim generated by slic3r is not holding the piece on the bed, the object curls and you can see that the object has detached from the brim

lordofhyphens, 2018-07-11:
Also a dupe of #4174 :) Looks like we can't get around a config option for this (which is something I was *really* hoping to avoid).

closed 2018-07-11 by lordofhyphens