slic3r/Slic3r #4486: large dimension data: slic3r never stops

Mettlerfreund, 2018-07-12:
### Version
1.2.9 up to newest version
last test with 1.3.1-dev (Slic3r-master.2018.07.12.1773.5dbfc10.64bit)

### Operating system type + version
windows 10 and linux alike

### Behavior
In order to slice 3D-objects with large dimensions (see #4478 and #2853) using the commandline you can downscale your object using ´--scale 0.01´ multiplying every coordinate value of the 3D-object file in this case by 0.01 prior to slicing. However this leads to Slic3r never ending its calculation for coordinate values around 100'000 ( scaling has to be around ´ --scale 0.01´ to have it work).

@lordofhyphens suggested: I suspect it's down in admesh or TriangleMesh (or the Perl conversion methods).

#### on Windows:

>´slic3r-console --scale 0.1 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 10_000.stl´
´=> Processing triangulated mesh´
´Use of uninitialized value in subroutine entry at C:/Users/sam/Desktop/temp/lib/Slic3r/Print.pm line 173.´
´Exporting to 10_000.svg...Done.´
´Warning: the supplied parts might not fit in the configured bed shape. You might want to review the result before printing.´

works fine

>´slic3r-console --scale 0.1 --layer-height 500 --first-layer-height 250 --nozzle-diameter 500 --export-svg 100_000.stl´
´=> Processing triangulated mesh´

never ends

#### STL/Config (.ZIP) where problem occurs
[issue.zip](https://github.com/slic3r/Slic3r/files/2189489/issue.zip)

Cheers

M

lordofhyphens, 2018-07-12:
I have some suspicions about this, particularly around the XS interface.

lordofhyphens, 2018-07-31:
I have a simple test written that I should be able to use for profiling, will check it when I have a workstation I can build at.

https://github.com/lordofhyphens/Slic3r/commit/c94c40e5c4809b2738a79ea3e389dd3a518cf7d0

Something else that has occurred to me is that the scaled coordinates may have gotten out of range of their datatypes.

lordofhyphens, 2018-07-31:
With the cppgui branch, TriangleMesh::slice() isn't the slow part here.

lordofhyphens, 2018-07-31:
@Mettlerfreund so testing with the C++ slicing interface, the slicing completes in under two minutes for `100_000.stl` (scaled 0.1 via  `TriangleMesh.scale()`).