slic3r/Slic3r #4488: stepping in one direction appears different than other

ruggb, 2018-07-13:
### Version
1.3.1 dev
Slic3r-master.2018.02.08.1593.05461b8.64bit

### Operating system type + version
W10 1803 17134-165

### Behavior
CoreXY machine
When printing diagonal lines, the width seems to be different when stepping the head toward a motor vs away from a motor. This is exhibited by a rough surface vs smooth surface.
Picture shows 1st layer with head moving away from right rear motor. Surface is very smooth. Rough surface was printed with head moving toward that motor. When 2nd layer was printed head moved toward left rear motor and was rough (less so because of diff layer height) and smooth on small corner when moving away from motor.
[image]
Because it occurs for either motor doing the line step, I can't rationalize a mechanical problem.

[DJI.zip](https://github.com/slic3r/Slic3r/files/2192766/DJI.zip)

lordofhyphens, 2018-07-13:
Given the nature of Gcode and the slicing toolpath, I can't rationalize a SW problem in the software. I also don't see this on my non-CoreXY. 

Your firmware would be the next SW component to investigate.

VanessaE, 2018-07-13:
Also, check your X carriage -- make sure it can't lean sideways.  In some bots, the belt will pull the carriage slightly off-center, making it lean left/right depending on motor direction.

ruggb, 2018-07-13:
further observation indicates that when the head is moving toward the motor that is stepping the line (not driving it forward) the surface is rough. When it is moving away from the motor it is smooth. 
The roughness, I believe, is because it is extruding too close to the previous line.
There is nothing in the model that asks for a change in the flow rate. It occurs for either motor when printing either diagonal. For coreXY on a diagonal, one motor drives it forward, one motor steps the lines. Also the advantage of a coreXY over an HBOT is that it does not apply tilting forces on the carriage. The anomaly is either extra extrusion or the line step in one direction is not the same as the other direction. IE, maybe 20 microsteps vs 15 microsteps. It is consistent which would not indicate mechanical looseness.
If it were extrusion, it should not change with direction. The only thing left is the slicer. BUT Marlin isn't home free either. If I can figure out from the gcode that the step distance is the same for either direction, then Marlin is on the line.
thanks

VanessaE, 2018-07-13:
That may be, but no matter the kinematics or belt path, you still have sideways forces on the carriage in the X direction.

ruggb, 2018-07-13:
but the belts are directly opposite so there is no tilting force

ruggb, 2018-07-13:
the gcode appears to step it a consistent 0.348/9 mm per step it either x or y direction so maybe it is not slic3r and is Marlin. There is currently a stepper timing issue and this may have something to do with it.

lordofhyphens, 2018-07-13:
@ruggb I am going to close this as "Not a bug" for now. If you find more evidence that points at something in Slic3r, I will reopen.

closed 2018-07-13 by lordofhyphens