slic3r/Slic3r #4489: Details within inner cylinder not printed

peterthevicar, 2018-07-18:
### Version
1.2.9

### Operating system type + version
Xubuntu 18.04

### Behavior
* Object with spokes going through a central cylinder does not print the part of the spokes within the central cylinder
* Load test.obj, compare 3D view with Layers view
* Printed layers should match the object
* 3D view is correct but printed layers omit the central part of the spokes. 
* Strangely, although slic3r reports no errors with the obj file (from FreeCAD) if you pass the obj file through the netfabb fixer the resulting file works as expected.

#### STL/Config (.ZIP) where problem occurs
Attached
[config-and-obj.zip](https://github.com/slic3r/Slic3r/files/2207079/config-and-obj.zip)

### Screenshots
[image]
[image]

lordofhyphens, 2018-07-18:
1.3.0 is out, by the way: https://github.com/slic3r/Slic3r/releases/tag/1.3.0

Please check with it.

I'd be willing to believe that Slic3r just didn't detect the model as garbage.

peterthevicar, 2018-07-19:
Thanks for the reply. I'm puzzled if the model is garbage how the 3D view works correctly. The problem only shows up in the layers view.
The 1.3.0 AppImage crashes shortly after startup on Xubuntu 18.04 so I built 1.3.0 from the sources; a couple of warnings but no errors. The result is just the same though. This screenshot is from 1.3.0, still no spokes inside the central cylinder.
[image]

peterthevicar, 2018-07-19:
I've done a bit more playing with this. If you load the spoked wheel into slic3r and then use "Load Part" to add the hub, you get the same fore-shortening of the spokes. Attached are the two obj files separately. I made the hub a lot bigger and it still has no spoke ends within it. A possible clue is that the hub, when loaded on its own, has some strange grey chords showing in the 3D view, perhaps indicating it's not a very well formed object?

[Wheel-and-hub.zip](https://github.com/slic3r/Slic3r/files/2209433/Wheel-and-hub.zip)

Dj-EKI, 2018-07-29:
Your Mesh overlaps other Meshes. Thats not good.
[image]

peterthevicar, 2018-08-02:
The part was formed using a Union operation in FreeCAD, sounds like a bug at that end. At the slic3r end I wonder why one part of the software (3D view) recognises correctly what the model is trying to do and the other part (slicer) does not? Is an overlapping mesh something that could easily be spotted and then either reported or repaired?

peterthevicar, 2018-08-02:
I've found what happened. In fact I DIDN'T perform the Union in FreeCAD, I simply selected the two parts and exported them as one .obj file. If I do a Union and then export the new object there is no mesh overlap and slic3r handles it fine.

lordofhyphens, 2018-08-02:
So may I close this ?