slic3r/Slic3r #4492: Q:

boelle, 2018-07-28:
### Version
1.41.0-alpha2+win64

### Operating system type + version
win10

### Behavior
* _Describe the problem_

i used to use version 1.39.2 and was very happy with it

i use it together witn my mendel90 and marlin bugfix 2.0.x

then the youtuber cnc kitchen showed the gyroid infill and i found out that i have to update, no big deal i thought, but i was wron

in version 1.39.2 i pretty much just copied the settings etc for the Mk2s and changed the start gcode, that was it

but in 1.41 it adds a section in the start with feedrates etc and then every now and then a progress code

now my printer skips steps because it moves to fast and i just cant figure how to get rid of that section in the start. i filled in the machine limits with just zero but then it moves so slow i cant see it moves

i even tried to create a custom setup from scratch but the same thing happend

any pinters or suggestions are welcome... and yes i know i dont have a prusa printer but it like the software :-/

xoan, 2018-07-28:
You are in the wrong place: if you use the Pruesa Edition of Slic3r, you should report issues in https://github.com/prusa3d/Slic3r

boelle, 2018-07-28:
upppps

let me copy over the issue first and then i will close it

my mistake... blind old man here :-(

closed 2018-07-28 by boelle

lordofhyphens, 2018-07-28:
@boelle if you can reproduce with a build from https://dl.slic3r.org/dev/ I can also take a look at this.

lordofhyphens, 2018-07-28:
1.3.1-dev has the gyroid infill too :)

boelle, 2018-07-28:
hmm... the tabs do not show up and each time i click on a menu i get the spinning wheel and "not responding" in the title bar

i do run the latest of win10 64 bit (danish)

lordofhyphens, 2018-07-28:
Open a profile to edit.

You may want to go to File->Preferences and change the editor mode to tabs

boelle, 2018-07-29:
but that assumes i can get to the menu

all i get is the spinning wheel and the "not responding" in the title bar

boelle, 2018-07-29:
[image]

boelle, 2018-07-29:
like this:
[image]

lordofhyphens, 2018-07-29:
Click on the gear icon next to the drop down.

boelle, 2018-07-29:
can you explain why i get the wheel icon so much and i also see a lot of "not responding" ?

lordofhyphens, 2018-07-29:
I don't know currently. Compare with 1.3.0 (see https://github.com/slic3r/Slic3r/releases/tag/1.3.0 )

boelle, 2018-07-29:
oh well, maybe the next version runs better, i thought there was some windows stuff that was missing