slic3r/Slic3r #4493: Compiling errors while using libslic3r in visual studio in c++ program

Chint27, 2018-07-28:
First of all,thanks..
I am beginner in C++. As I want to use libslic3r in my program , I built it using Cmake based on instruction given on Slic3r website(built from source). I was successful in building it using MinGW compiler  and Cmake (ignored WX not found error)and got libslic3r.a files. Now in order to link in visual studio,I linked required directories and boost and its corresponding libraries in Visual Studio Community2015 . I tried to use libslic3r and tried to compile my program.  When I tried to compile it gave me errors in polygon.h file and expolygon.h file. Can you please help me.. I have attached snapshot of the errors I got.
[image]

lordofhyphens, 2018-07-28:
Looks like you don't have Boost configured in your project. You also wouldn't be able to link the .a files produced by MINGW with VS (so far as I am aware).

I do not use Visual Studio at all myself.

lordofhyphens, 2018-07-28:
Look online for using cmake with Visual studio. 

As for using libslic3r, take a look at the tests in src/test/libslic3r  (particularly the init_print method).

Curre

Chint27, 2018-07-28:
@lordofhyphens , I am thankful for your advice.. I have to built libslic3r with cmake and visual studio(not MinGW) ,  have I  understood it properly?? In my visual studio, i also included boost libraries and its directories .

lordofhyphens, 2018-07-28:
I'm going to install VS2017 Community and see if I can reproduce the problem.

The error messages, point to something going on with your system, though, or VS2015.

Chint27, 2018-07-28:
@lordofhyphens  , My system is Windows 10 , 64 bit ...

Chint27, 2018-07-29:
@lordofhyphens  did you succeed in building libslic3ror using it in visual studio community 2017??

Chint27, 2018-07-30:
@lordofhyphens , I tried to build it with visual studio and cmake , when I built with visual studio 2015 community , it failed to built few directories like libslic3r (it couldnot generate trianglemesh.cpp object file) but built other directories and I also tried with visual studio community 2017, there also it failed..

lordofhyphens, 2018-07-30:
@Chint27 Slic3r on Windows is built with MinGW's gcc compiler (specifically [mingw-w64](https://mingw-w64.org/doku.php)).

Chint27, 2018-07-30:
@lordofhyphens , if i want to use only libslic3r , how to built in Visual Studio. I dont want entire slic3r, just libslic3r

lordofhyphens, 2018-07-30:
@Chint27 You want the CMakeFiles in the `./src` directory (you should update your checkout by the way if you haven't in a week or so).

Build just the `libslic3r` target. See this [example from stack overflow](https://stackoverflow.com/questions/25896657/what-does-target-option-mean-in-cmake).

As I said, I do not use Visual Studio at all so I can't give you step-by-step instructions for that.

lordofhyphens, 2018-07-30:
That is, I will need to set up a build environment myself first before I could properly advise you.

Chint27, 2018-08-06:
@lordofhyphens , Hey finally I used Prusa3d Slic3r and followed instruction to compile with visual studio 2013 on its wiki and obtained libslic3r.lib and other .lib files successfully. Now I am trying to link these files in my program in visual studio2013... Is it possible to link it and use  libslic3r ?, as I am getting few linking errors....

Chint27, 2018-08-10:
@lordofhyphens , I am very thankful for the steps to build libslic3r. I will follow this. Once again thank you!!!!

lordofhyphens, 2018-08-10:
@Chint27 You may want to modify the CMake file and add the option to build libslic3r as a DLL. If you get that working I'd probably accept a pull request to add it to Slic3r.

closed 2018-08-10 by Chint27