slic3r/Slic3r #4496: Anchorage of top & bridge infill : new parameters?

supermerill, 2018-07-30:
Hi
There are been some issue where user wanted to tune these parameter:
https://github.com/prusa3d/Slic3r/issues/668
https://github.com/prusa3d/Slic3r/issues/932
Currently, it's a #define to 3mm
What most of people want is reducing the anchorage of top infill to near 0 and let the anchorage of the bridge infill to 3.

I've created two new parameter in my fork.
These parameter are really needed or just a waste of gui space?
Maybe only the "top" one?

tagging @VanessaE (what is willmore github name?)

Duckle29, 2018-07-30:
I agree having this configurable would be great. I've wasted a lot of plastic on models that really didn't need it because of this.

djdelorie, 2018-07-30:
I think the code could be smarter about detecting and computing needed anchorage, without a GUI change.  I suspect it's a lot of work though.

* Consider support overhang limit for the "inside" face of shells
* Extend top/bridge to next infill support, not some arbitrary amount

supermerill, 2018-07-30:
> Consider support overhang limit for the "inside" face of shells

sorry but i don't understand

> Extend top/bridge to next infill support, not some arbitrary amount

it extends into the sparse infill, in (almost) every case.

bubnikv, 2018-07-31:
> it extends into the sparse infill, in (almost) every case. Or I don't

>  it extends into the sparse infill, in (almost) every case. Or I don't
understand.

It is this quite natural proposal:

https://github.com/prusa3d/Slic3r/issues/569

supermerill, 2018-07-31:
Oh, i see.
That's why I created the "dense" infill (https://github.com/prusa3d/Slic3r/issues/938), to avoid working on that :-p .

More seriously, maybe if I split the sparse surfaces on the makefill() pre-processing steps if a dense layer is over them, and asking the infill pattern to ensure it draw a near complete line over the edge, it should do the trick.

supermerill, 2018-08-09:
some progress: https://github.com/supermerill/Slic3r/wiki/Dense-infill-under-solid-layer

https://camo.githubusercontent.com/c0fb77e2f6f0a24d1843d0ad4a12c56181a5aab0/68747470733a2f2f692e696d6775722e636f6d2f464843433231672e6a7067

jkuusama, 2019-02-24:
I added another example to [prusa3d#569](https://github.com/prusa3d/Slic3r/issues/569). I think this is serious, as this is one, if not the(!) cause for the layer shifts.