slic3r/Slic3r #4497: a Numerical Bug in the C++ library libslic3r about the GCodeWriter::lift() function

Enyi-Tang, 2018-07-30:
### Version
1.3.0-387-g34e633a (The latest version of git-master on July 29, 2018. The full hash is '34e633a1c07901d278ef13474b08316c3aa196f1')

### Operating system type + version
Ubuntu 16.04 LTS x64 Desktop

### Behavior
The C++ library libslic3r(core of Slic3r) has a numerical bug that causes GCodeWriter::lift() not to work anymore even after calling GCodeWriter::unlift(). The bug can be reproduced by the following code:

``` C++
/// bug_lift.cpp
#include <algorithm> 
#include <iostream>
#include <iostream>

#include "TriangleMesh.hpp"
#include "libslic3r.h"
#include "Point.hpp"
#include "GCodeWriter.hpp"

using namespace Slic3r;

std::string BreakLineIfEmpty(std::string s) {
  if(s.length()==0){
    return "\n";
  }else{
    return s;
  }
}

int main() {
  GCodeWriter* writer = new GCodeWriter();
  writer->config.set_defaults();  
  writer->config.load("config_load.ini.txt"); // retract_lift = 1.5
 
  std::vector<unsigned int> extruder_ids; 
  extruder_ids.push_back(0);
  writer->set_extruders(extruder_ids);
  writer->set_extruder(0);

  std::cout<<"to_z 0992.0:"<<BreakLineIfEmpty(writer->travel_to_z(9007199254740992)); 
  std::cout<<"lift       :"<<BreakLineIfEmpty(writer->lift());  // The first lift is correct.
  std::cout<<"to_z 0993.5:"<<BreakLineIfEmpty(writer->travel_to_z(9007199254740993.5));  
  std::cout<<"unlift     :"<<BreakLineIfEmpty(writer->unlift());
  std::cout<<"lift       :"<<BreakLineIfEmpty(writer->lift()); // The second lift does not work even after unlift().

  delete writer;
  return 0;
}
```

I attached the configuration file "config_load.ini.txt" with the related configurations:
retract_lift = 1.5
retract_lift_above = 0
retract_lift_below = 0
[config_load.ini.txt](https://github.com/slic3r/Slic3r/files/2241032/config_load.ini.txt)

* _Steps needed to reproduce the problem_
Just compile the above code and link it to the libslic3r (core of Slic3r). On my computer, I build it with gcc 5.4.0 by the following command: (It works well.)
`g++ -o bug_lift bug_lift.cpp -std=c++11 -I../xs/src -I../xs/src/libslic3r -pthread ../build/liblibslic3r.a ../build/libadmesh.a ../build/libBSpline.a ../build/libclipper.a ../build/libexpat.a ../build/libpolypartition.a ../build/libpoly2tri.a ../build/libZipArchive.a /usr/lib/x86_64-linux-gnu/libboost_system.a /usr/lib/x86_64-linux-gnu/libboost_thread.a /usr/lib/x86_64-linux-gnu/libboost_filesystem.a /usr/lib/x86_64-linux-gnu/libboost_chrono.a /usr/lib/x86_64-linux-gnu/libboost_date_time.a /usr/lib/x86_64-linux-gnu/libboost_atomic.a`

* _Expected Results_
GCodeWriter should export the gcode string at the lift() action immediately after the unlift() action. Specifically, when the configuration of "retract_lift =1.5", the program should print to the console as follows:

```
to_z 0992.0:G1 Z9007199254740992.000 F7800.000
lift       :G1 Z9007199254740994.000 F7800.000
to_z 0993.5:
unlift     :
lift       :G1 Z9007199254740996.000 F7800.000
```

* _Actual Results_
However, the lift() action immediately after the unlift() action is ignored. The output screenshot is as follows:
[image]
If we add more lift() and unlift() after that, the program will ignore them all.

* _Bug tracing_
We trace the bug to a few numerical calculation in the library. Hope the developer can confirm the bug. We would like to provide some fixing advice.

#### Config where problem occurs
[config_load.ini.txt](https://github.com/slic3r/Slic3r/files/2241032/config_load.ini.txt)

lordofhyphens, 2018-07-30:
So this only appears at a ridiculously large Z height?

lordofhyphens, 2018-07-30:
@Enyi-Tang does  https://github.com/lordofhyphens/Slic3r/blob/cppgui/src/test/libslic3r/test_gcodewriter.cpp appropriately describe the test case?

Enyi-Tang, 2018-07-30:
I believe the bug appears when Z and GCodeWriter::_lifted value appears in very different magnitude. I am not sure if a big GCodeWriter::_lifted with a small Z, or a small Z with a tiny GCodeWriter::_lifted will also cause the problem.

Enyi-Tang, 2018-07-30:
@lordofhyphens When I check the test case you mentioned, I am not sure if the unlift() action should also output a gcode string itself. It is about the line 29 of https://github.com/lordofhyphens/Slic3r/blob/cppgui/src/test/libslic3r/test_gcodewriter.cpp: 
"REQUIRE(writer.unlift().size() > 0);"

lordofhyphens, 2018-07-30:
@Enyi-Tang checkout cppgui and run the test. I'd like to have this fail in a reproducible way before spending effort trying to trace code and fix it. 

If I've gotten the test incorrect, feel free to fix it and open a PR against cppgui for it (I'll merge it into master once we know it is a valid test). 

The test as it's written is passing on the build server with no code changes.

lordofhyphens, 2018-07-30:
@Enyi-Tang Yeah, I just realized that: unlift() doesn't do anything because there is no Z change. I've fixed the test.

Enyi-Tang, 2018-07-30:
@lordofhyphens Could you check the test case below? See if it can pass on the build server?

```C++
#include <catch.hpp>
#include <memory>

#include "GCodeWriter.hpp"
#include "test_options.hpp"

using namespace Slic3r;
using namespace std::literals::string_literals;

SCENARIO("lift() is not ignored after unlift() at large values of Z", "[!mayfail]") {
    GIVEN("A config from a file and a single extruder.") {
        GCodeWriter writer;
        auto& config {writer.config};
        config.set_defaults();
        config.load(std::string(testfile_dir) + "test_gcodewriter/config_lift_unlift.ini"s);

        std::vector<unsigned int> extruder_ids {0};
        writer.set_extruders(extruder_ids);
        writer.set_extruder(0);

        WHEN("Z is set to 9007199254740992") {
            double trouble_Z1 = 9007199254740992;
            config.retract_lift.values[0] =1.5;  // not sure if it is correct.
            double trouble_Z2 = 9007199254740993.5;
            writer.travel_to_z(trouble_Z1);
            AND_WHEN("GcodeWriter::Lift() is called") {
                REQUIRE(writer.lift().size() > 0);
                AND_WHEN("Z is moved post-lift to the same delta as the config Z lift") {
                    REQUIRE(writer.travel_to_z(trouble_Z2).size() == 0);
                    AND_WHEN("GCodeWriter::Lift() is called after GCodeWriter::Unlift()") {
                        REQUIRE(writer.unlift().size() == 0);
                        THEN("GCodeWriter::Lift() emits gcode.") {
                            REQUIRE(writer.lift().size() > 0);
                        }
                    }
                }
            }
        }
    }
}
```

lordofhyphens, 2018-07-30:
Ah, there's the test failure: https://travis-ci.org/lordofhyphens/Slic3r/jobs/409938331#L5779

lordofhyphens, 2018-07-30:
@Enyi-Tang just make a pull request against my fork and the build server will run it.

ledvinap, 2018-07-30:
@lordofhyphens : Is this really worth investigation? `9007199254740992` is 2^53, at edge of IEEE double representation (next representable number is 9007199254740994).

`long double` can be used if someone wants to print earth at nanometer precision, but they should have enough resources to do the conversion themselves ...

Enyi-Tang, 2018-07-30:
I do not know if it is worth for investigation. From our tracing, the problem occurs in the function of travel_to_z:
```
double nominal_z = this->_pos.z - this->_lifted;
this->_lifted -= (z - nominal_z);
```
And it can be fixed with:
```
this->_lifted== this->_pos.z - z;
```

lordofhyphens, 2018-07-30:
I have it tagged as "mayfail" so you'll need to look at the result.

ledvinap, 2018-07-30:
@Enyi-Tang : Using few times larger `z` (and searching a bt for 'problematic' value) will trigger this problem even with your version ...
Both versions are mathematically equivalent(but not numerically due to rounding), but existing version matches code comment ...

lordofhyphens, 2018-07-30:
@ledvinap yeah it being ridiculously large as a pointf was something that stuck out to me. It's something that I'd probably leave a test in for (with the behavior marked as shouldfail), but you're right, it's on the edge of the underlying datatypes and I can't find a justification for Slic3r to change out to `long double`.

Enyi-Tang, 2018-07-30:
@ledvinap I see. Changing numerical code really affects the maintainability.

lordofhyphens, 2018-07-30:
I've added some tests to cover expected behavior at more normal behavior, the failure case shown occurs with Z values larger a few times the distance from the earth to the sun.

lordofhyphens, 2018-07-30:
Closing as the failure condition should not be encountered in terrestrial use of Slic3r.

closed 2018-07-30 by lordofhyphens

lordofhyphens, 2018-07-30:
@Enyi-Tang I will happily accept PRs with more robust tests for Slic3r in the future :)

Enyi-Tang, 2018-07-31:
@lordofhyphens All right, I will try my best.