slic3r/Slic3r #4498: Does not find all OctoPrint(-compatible) servers on the network

probonopd, 2018-07-30:
Slic3r does not find all OctoPrint(-compatible) servers on the network (in my case, I am trying to make it work with https://github.com/probonopd/WirelessPrinting). I think the fault lies within `Net::Bonjour`. I have tried to contact the author of `Net::Bonjour` but unfortunately the published mail address seems not to be reachable anymore.

Please let me know how to best debug this. Is there any type of output (e.g., Wireshark) that would be useful?

I have written a tiny example to show the problem. If you run:

```
sudo apt install npm
sudo npm install bonjour
node advertise.js
```

with the following advertise.js file:

```
var bonjour = require('bonjour')()

bonjour.publish({ name: 'OctoPrint instance on MACHINE', type: 'octoprint', port: 80, txt: ({"path": '/', "api": '0.1', "version": '1.2.10'}) })

bonjour.find({ type: 'octoprint' }, function (service) {
  console.log('Found an OctoPrint server:', service)
})
```

Then run the following discover.pl while the above advertise.js is running:

```
use Net::Bonjour;

my $res = Net::Bonjour->new('octoprint');
$res->discover;

foreach my $entry ( $res->entries ) {
        printf "%s %s:%s\n", $entry->name, $entry->address, $entry->port;
}

```

`Net::Bonjour` _fails_ to find the service. This is on `xubuntu-18.04-desktop-amd64.iso`.

lordofhyphens, 2018-07-30:
I'll have to think more on that; I don't have much experience with using Bonjour or that library.

It looks like Net::Bonjour is pure perl, so on a specific system you can modify the copy in local::lib and start snooping around in there.

probonopd, 2019-03-02:
The 1.3.1-dev 05a2f90 AppImage does _not_ recognize my device. I had the exact same issue in Slic3r PE where it since has been [fixed](https://github.com/prusa3d/Slic3r/issues/1864#issuecomment-468713447).

They have dropped the Perl implementation and are using their own in C++.