slic3r/Slic3r #4499: Download folder seems to be full.

mag00-br, 2018-07-30:


lordofhyphens, 2018-07-30:
I'll have to sort it out tonight; thanks for the heads-up.

lordofhyphens, 2018-08-01:
@mag00-br builds are available again.

lordofhyphens, 2018-08-01:
@alexrj the drive space seems to be taken up by something else that isn't in the folder.

scott0122, 2018-08-08:
Slic3r dev build downloads still not working.

lordofhyphens, 2018-08-08:
@scott0122 Out of space; @alexrj has to sort out the server (I do not have sufficient access rights).

lordofhyphens, 2018-08-08:
I have kicked off builds and the space situation should have been resolved. 

Check in about an hour.

lordofhyphens, 2018-08-08:
Windows and OSX builds look fine. Something looked off about the Linux build sizes, so I will look at that later.

lordofhyphens, 2018-08-20:
There's the Linux build (broken for unrelated reasons). Closing.

closed 2018-08-20 by lordofhyphens